ros2/rclcpp #10: Move rmw_implementation to the rmw repo

wjwwood, 2015-03-05:
Currently this repository contains the `rmw_implementation` package which provides an interface to list and select an implementation for the `rmw` API. It also sets the default implementation. We agreed that it should be in the `rmw` repository as a peer to the `rmw` package instead.

Acceptance Criteria:

- [ ] `rmw_implementation` package moved
- [ ] git history preserved
- [ ] ros2 is still building after the move

dirk-thomas, 2015-03-07:
It was a bad idea to move it into the `rmw` repository since that generates a bad dependency chain:

    rmw_implementation (in rmw) -> rmw_opensplice_cpp (in rmw_opensplice) -> rwm (in rmw)

That was actually the reason why it was not in the rmw repo before.

dirk-thomas, 2015-03-07:
[image]

dirk-thomas, 2015-03-07:
For the same reason `rosidl_default_generators` should be moved out of the `rosidl` repo.

wjwwood, 2015-03-07:
What's wrong with the dependency chain? Just that you can't easily get `rmw` without the `rmw_implementation` package?

dirk-thomas, 2015-03-07:
Two repositories depending on each other is a not desired circular dependency.

wjwwood, 2015-03-07:
Ok, I agree, then what should we do? As I see it we can either move `rmw_implementation` to a different existing repository or move it to its own repository.

I am leaning towards its own repository, but it would be nice to not have yet another repository to deal with.

dirk-thomas, 2015-03-07:
I share your concern to keep the repo count low. But I don't think it fits well into any of the existing repos and therefore I also lean towards a new separate repo.

wjwwood, 2015-03-31:
What should we call this repository? Just `rmw_implementation`?

dirk-thomas, 2015-03-31:
I don't have a better suggestion and would therefore suggest to keep the name as-is.

wjwwood, 2015-03-31:
The new repository is here: https://github.com/ros2/rmw_implementation

wjwwood, 2015-03-31:
Ok, I've opened pull requests to address the move. Please review.

dirk-thomas, 2015-03-31:
lgtm

closed 2015-03-31 by wjwwood