ros2/rclcpp #35: Add support for a service response callback with request payload embedded

tfoote, 2015-05-26:
This was identified as potentially valuable during the parameter API review. 

The use case is that there are a sequence of queries in a request, the response objects can be sorted in the same order as the requests, instead of requiring the response to be entirely self describing.

garyservin, 2015-10-20:
@tfoote can you add more info for this issue? I'm feeling a bit out out of context here. Thanks!

tfoote, 2015-10-21:
This was mostly in the context of an async service. Where at the time of response the initial request may be out of scope. In that case the response could be coupled with the request. 

For example [here]( https://github.com/ros2/examples/blob/master/rclcpp_examples/src/services/add_two_ints_client_async.cpp#L35) you could print both the request and the result instead of just the result. This saves the user from having to book keep which request is associated with which response if they're running multiple in parallel, which is not represented in the simple example.

closed 2015-11-19 by esteve