ros2/rclcpp #52: Subscriptions are not triggered sometimes

esteve, 2015-06-30:


gerkey, 2015-07-07:
Happens in OpenSplice and Connext, so it's likely in our code somewhere.  Consider writing a test that reproduces the problem.  Also walk somebody else on the team through the problem.

dirk-thomas, 2015-07-30:
The same fix as implemented for subscriptions (https://github.com/ros2/rmw_connext/pull/62 https://github.com/ros2/rmw_opensplice/pull/57) needs to be applied for server and client.

dirk-thomas, 2015-08-04:
We will try to write a test which fails before and passes afterwards. E.g. similar to the subscription: calling two services, spin and likely the second request is never handled.

dirk-thomas, 2015-10-15:
I implemented the test in ros2/system_tests#45.

dirk-thomas, 2015-10-15:
@esteve Can you please test the original problem related to parameter events again with the patches being merged and either close this ticket or fill in more information what still doesn't work.

closed 2015-10-15 by esteve