ros2/rclcpp #86: Pub/sub across different nodes in same executor does not work

jacquelinekay, 2015-08-13:
Branch `multiple_nodes` in `system_tests`, package rclcpp, illustrates this bug:
https://github.com/ros2/system_tests/tree/multiple_nodes
(^ the test needs ros2/rclcpp/pull/84 for `Executor::spin_some`)

Case 1 fails:
node1 and node2 are both added to one executor.
node1 publishes "foo", node2 subscribes to "foo"
node2 publishes "bar", node2 subscribes to "bar"
Both publishers publish 5 times.
0/5 messages are received for both subscribers.

Case 2 passes:
one node publishes "foo", "bar", subscribes to "foo" and "bar"
Both publishers publish 5 times.
5/5 messages are received for both subscribers.

Case 3 passes:
node1 and node2 both added to one executor.
node1 publishes "foo", subscribes to "foo"
node2 publishes "bar", subscribes to "bar"
Both publishers publish 5 times.
5/5 messages are received for both subscribers.

wjwwood, 2015-08-13:
Is this consistent across vendors?

jacquelinekay, 2015-08-13:
Actually, it looks like it's specific to Connext. I should probably move
this issue from rclcpp to rmw_connext, then.

closed 2015-08-13 by jacquelinekay