ros2/rclcpp #104: Real-time safe services

jacquelinekay, 2015-09-08:


BobDean, 2015-09-12:
there are a couple of open items with out descriptions. If those could be populated, even with something simple, it would help understand what is being worked on but is not yet in the public repos.

gerkey, 2015-09-12:
@BobDean: FYI, everything we're doing is in public repos, but perhaps on branches.  But I take the point about having more descriptive tickets, as it gives more information about what remains to be done (which will hopefully make it easier for external contributors to pitch in).

BobDean, 2015-09-12:
@gerkey my "public" comment was due to reading tickets which seemed to relate to something on the developers machine, which I did not see when browsing the github source. So maybe the ticket progress/resolution had not been pushed to github yet. Which is fine, that's how software development goes. The description would let me fill in the gap while browsing.

jacquelinekay, 2015-11-10:
I have a branch abstracting out allocations in services and clients, but I'm blocked by an issue with allocations from within the stdlib (`std::promise`). I've asked it on Stack Overflow:

http://stackoverflow.com/questions/33638134/stdpromise-with-custom-allocator-appears-to-use-global-new

jacquelinekay, 2015-11-11:
It turns out the issue with `promise` will be fixed in gcc 5. Unless it gets backported to gcc 4.8 or 4.9, that means we won't be able to use `std::promise` for RT services and clients until we start targeting 16.04.