ros2/rclcpp #109: parameter client question

BobDean, 2015-09-12:
so...
node caches parameters locally.
but sends them out to the parameter_event channel... so if all nodes are up, all nodes can get the events.
but it seems the parameter server does not listen to this channel, it relies on a service to set params in its cache.  the server requires a separate api for access (parameter_client). 
are the parts just not hooked up yet? or am i missing something?

tfoote, 2015-09-14:
Hmm it looks like we've overlooked the parameter design document. 

I'm working up a draft design document that I hope can answer most of your questions. https://github.com/ros2/design/issues/57

tfoote, 2015-09-15:
@BobDean I've put together a first draft of a parameter design dock. please follow up on ros2/design#58

closed 2015-09-15 by tfoote