ros2/rclcpp #114: Provide CMake infrastructure for creating a node

wjwwood, 2015-09-16:
This is a spin off from https://github.com/ros2/ros2/issues/55.

Basically we need to add some lubrication to the process of making a node which can be loaded as a library into a container or run as a stand-alone executable. This is what I imagine it would look like:

```cmake
rclcpp_add_node(my_node src/my_node.cpp)
target_link_libraries(my_node ...)
ament_target_dependencies(my_node "rclcpp" ...)
...
```

The `rclcpp_add_node` macro would create a target `my_node` which builds a shared library, and also creates a target which builds an executable (name of this target is an implementation detail) with the output name `my_node`. This allows the user to link extra libraries and what not to the library containing the node. It would also register this node in the ament index.

The output files will be `lib/libmy_node.so`, `bin/my_node`, and `share/ament_index/resource_index/nodes/pkg_name/my_node` (maybe just `share/ament_index/resource_index/nodes/pkg_name` with the `my_node` in the contents of that file).

The user's code is compiled into `lib/libmy_node.so`.
The executable is a boilerplate main which can load and run the shared library in stand alone mode or, with a cli switch, it can tell a remote container to load and run the library, acting as a surrogate until the node is removed from the container.

The infrastructure has been prototyped here: https://github.com/ros2/demos/pull/18

Making all of this work will require us to figure out the issues surrounding multi vendor support in https://github.com/ros2/rclcpp/issues/48

Karsten1987, 2018-02-26:
@wjwwood same as https://github.com/ros2/rclcpp/issues/115#issuecomment-368631239

wjwwood, 2018-02-26:
I would say no. We still do not have CMake infrastructure to let users easily create node components. And I still think we should.

jacobperron, 2019-06-07:
I've proposed this for a Dashing patch release since it enables https://github.com/ros2/demos/issues/325.

wjwwood, 2020-03-19:
Closed by https://github.com/ros2/rclcpp/pull/784?

closed 2020-03-19 by wjwwood