ros2/rclcpp #115: Create a generic node container

wjwwood, 2015-09-16:
This is a spin-off from: https://github.com/ros2/ros2/issues/55

We need a generic program which can be run and then remotely instructed to load and execute a node.

This program should probably:
- Allow configuration of the internal executor.
- Provide ROS Services for loading and running a node.

This is not really useful until we have the ability to make, find, and dynamically load nodes, see:
- https://github.com/ros2/rclcpp/issues/114
- https://github.com/ros2/demos/pull/18

This is also blocked by decisions that need to be made in: https://github.com/ros2/rclcpp/issues/48

Karsten1987, 2018-02-26:
@wjwwood Can this be closed given that we have the composition demos?

wjwwood, 2018-02-26:
No, the demo just demonstrates the technology, it doesn’t provide a general purpose executable that people can rely on going forward. It wouldn’t be a good idea for people to build robot launch files that launch critical executables from a demo package.

rotu, 2019-07-02:
@wjwwood How about now that `rclcpp_components component_container` exists?

ZhenshengLee, 2023-10-31:
Canbe closed? @wjwwood

mjcarroll, 2023-10-31:
Yes, I think we are good here.

closed 2023-10-31 by mjcarroll