ros2/rclcpp #150: Services using 100% of CPU

esteve, 2015-11-12:
Dave Sandage reported this on the NG SIG:

```
I've just started playing with the ROS2 alpha2 release, and was experimenting with some samples.  I wrote a node to act as a service server, modeling it after the add_two_ints_server example.  It all works, but after Ive made my first call to the service, the server node pegs the CPU at 100%.  add_two_ints_server does this also.

At the end of my node's main() function, I have the following loop:

while (rclcpp::utilities::ok())
{
    executor.spin_node_once(node, std::chrono::milliseconds(1000));
}

After the first call to the service, this method returns immediately every time, leaving the node in a tight loop.  Upon further debugging in rclcpp, it appears that in executor::execute_next_ready_executable(), get_next_service always returns a non-null value.  Is there something in my service handler that needs to be done to remove the service from showing always ready to execute?  I've modeled mine exactly on the add_two_ints_server (and it does the same thing).
```

firesurfer, 2016-02-03:
Hey,
is there already a fix available for this issue?

dirk-thomas, 2016-02-05:
@firesurfer I am not sure if the problem still exists or has been fixed already. If it would have been reported for the alpha1 I would have assumed it has been addressed by these (ros2/rmw_opensplice#60 ros2/rmw_connext#66). It would be great if you could create a simple example which demonstrates the problem.

firesurfer, 2016-02-08:
I don't think that it is this specific problem. It seems more like that a huge number of created publishers/subscribers + services + parameterclients create a huge cpu load even if they are idle. 
I'm trying to reproduce this phenomena in a simple example but can't get it to work at the moment. 

By profiling my application with callgrind it seems that it spends most of the time in the rclcpp::executor::Executor::wait_for_work method called from the executor.
My application creates about 40 topics with about 30 publishers and 10 subscribers, about 40 parameter clients which access a single parameter service and 40 custom services. All of them are idle at the moment after creating them. Only the parameter clients are used once for accessing the central parameter service that runs in another process.

firesurfer, 2016-02-09:
I found the problem. It depends on the interface the dds implementation chooses. In my case I had my application run on my laptop that is connected via wireless lan, but tried to create an example for demonstration on my desktop, which is connected via lan. 

Nonetheless I find it a bit strange that using a wirelesslan connection results into a high cpu load.

dirk-thomas, 2016-02-13:
We have just fixed another case where `rclcpp` would start busy looping (see ros2/rmw_opensplice#112). It would be great if you could try the latest versions from all repos and comment back if it helps for this case.

firesurfer, 2016-02-14:
I pulled the latest version from the master branch. This issue is still existing, but I can't really test it because of the exception I mentioned in https://github.com/ros2/rclcpp/pull/193

firesurfer, 2016-02-15:
By replacing the exception with a return I was able to test. The average cpu load in idle is much less than before. Around 20% on an older i7 cpu, same amount of publishers and so on as before. 

But it doesn't seem to depend on whether I use a wireless or a wired connection anymore. In both cases I got the same cpu load.

dirk-thomas, 2016-02-15:
While that sounds better 20% is still far away from idle. Can you either share an example which has that problem or debug locally to figure out where the CPU is being used in your case (at best using the debugger and identifying where it probably busy loops)?

firesurfer, 2016-02-15:
I will try tomorrow tomorrow to recreate the scenario my application creates in an example. I hope this will help tracking down this issue and #192

firesurfer, 2016-02-18:
Hi my example is in https://github.com/firesurfer/ros2_components_demo (the same example I already used for some other issues - please pull the newest version, I separated the ros2_components_demo and the ros2_components repo)

On my system opensplice dds automatically chooses wlan0. 
The demo_client takes about 90% cpu time (meassured by htop) 
the demo_server takes about 50 % cpu time
and the demo_parameter_server takes abot 2-3% cpu time
on a system running a Intel(R) Core(TM) i7-2670QM CPU @ 2.20GHz

dirk-thomas, 2017-01-03:
@firesurfer Looking at the changes in your repo I assume this has been fixed in the meantime. Can this be closed?

firesurfer, 2017-01-03:
Yes this is fixed and can be closed.

dirk-thomas, 2017-01-03:
Thanks for confirming.

closed 2017-01-03 by dirk-thomas