ros2/rclcpp #173: subscriber callbacks bound to class methods do not compile

tfoote, 2015-12-11:
It fails due to an inability to resolve an overloaded function. 

```
/home/tfoote/work/ros2/tf3/install/include/rclcpp/function_traits.hpp:51:85: error: decltype cannot resolve address of overloaded function
       typename function_traits<decltype( & FunctionT::operator())>::arguments>::type;
```

The full error is here: https://gist.github.com/tfoote/17222805a01e7c05c3f1

jacquelinekay, 2015-12-11:
Can you make this a test case in system_tests/test_rclcpp instead?

jacquelinekay, 2015-12-11:
Are you working on a fix for this right now? I will take a look if you're not actively working on it.

tfoote, 2015-12-11:
@esteve found a workaround to use a lambda function. I'll add a unit test for it too.

jacquelinekay, 2015-12-11:
You mean using a lambda function in the example? Or to make bound functions work as the input to create_subscription?

None of our tested rclcpp code so far uses `std::bind`. Any code that captures arguments does so using lambdas.

Playing around with this, I found another "workaround" that compiles:
```
std::function<void(const std_msgs::msg::String::SharedPtr)> callback = std::bind(&Chatter::chatterCallback, &chatter, std::placeholders::_1);
  auto sub = node->create_subscription<std_msgs::msg::String>(
    "chatter", callback, rmw_qos_profile_default);
```

Ideally we would support any valid callback signature for create_subscription, so I think we should do the necessary work to make your example compile.

esteve, 2015-12-11:
@tfoote could you have a look at this branch and see if it works? Thanks.

https://github.com/ros2/rclcpp/tree/std_bind_function_traits

esteve, 2015-12-11:
@tfoote I got your example compiling with my branch, I'll clean up the changes here and submit them for review.

tfoote, 2015-12-12:
@esteve I can confirm that your branch works for me, but I see it's not a cross platform solution unfortunately :-(

esteve, 2015-12-16:
@tfoote I got the workaround to compile on all three platforms in #183, I'll see if I can compile the `std::bind` test in ros2/system#88

tfoote, 2015-12-17:
fixed in #183

closed 2015-12-17 by tfoote