ros2/rclcpp #190: rate.sleep will sometimes return slightly early from a sleep

tfoote, 2015-12-18:
I was working on the unit tests with @gerkey for rate.sleep and expected the sleep to not return before expected. 

On repeated runs it had a jitter on the order of 10us. (The printout is in nanoseconds. and the expected duration is 100 ms

Console output of observed delta: https://gist.github.com/tfoote/0ec473b38bc517d12779

We added an epsilon to the unit tests for now (https://github.com/ros2/rclcpp/pull/182), but we should review our code to make it as consistent as with the assumptions of developers who are used to calling the standard sleep methods.

jacquelinekay, 2016-01-08:
Is the console output from Linux?

gerkey, 2016-01-08:
This problem was fixed in the test, in #191.

closed 2016-01-08 by gerkey

jacquelinekay, 2016-01-08:
But the issue that `Rate::sleep` wakes up 10 microseconds early still persists, right?

gerkey, 2016-01-08:
Nope, it was working properly the whole time. The bug was in the test, which caused the addition of the epsilon workaround mentioned here. But then the test was fixed and the workaround removed.