ros2/rclcpp #192: Segfault at:  spin_until_future_complete

firesurfer, 2016-01-30:
Short description of my program:
static method A: gets a node passed, accesses a parameter server via an AsyncParametersClient
returns some information gained from the parameter server

static method B:gets a node passed: accesses a service that is hosted somewhere.
returns some information gained from the service

Both methods have somewhere a call to rclcpp::spin_until_future_complete.

Example:
When I call method A, A returns succesfully.
When I call method B, B will give me a segfault

Everytime I'm using a the service call in a static method (that succeeds), I cant do a second service call or a call to a parameter server because of a segfault at:  rclcpp::spin_until_future_complete

gdb Backtrace:
```
#0  0x00000000006747b8 in std::__shared_ptr<std::__future_base::_State_baseV2, (__gnu_cxx::_Lock_policy)2>::operator-> (this=0x0) at /usr/include/c++/5/bits/shared_ptr_base.h:1055
#1  0x00007ffff201e263 in std::promise<std::shared_ptr<rcl_interfaces::srv::GetParameters_Response_<std::allocator<void> > > >::set_value(std::shared_ptr<rcl_interfaces::srv::GetParameters_Response_<std::allocator<void> > > const&) () from /home/firesurfer/workspace/ros2_ws/install/lib/librclcpp.so
#2  0x00007ffff201cfbb in rclcpp::client::Client<rcl_interfaces::srv::GetParameters>::handle_response(std::shared_ptr<void>&, std::shared_ptr<void>&) ()
   from /home/firesurfer/workspace/ros2_ws/install/lib/librclcpp.so
#3  0x00007ffff1fcc7f6 in rclcpp::executor::Executor::execute_client(std::shared_ptr<rclcpp::client::ClientBase>) () from /home/firesurfer/workspace/ros2_ws/install/lib/librclcpp.so
#4  0x00007ffff1fcc193 in rclcpp::executor::Executor::execute_any_executable(std::shared_ptr<rclcpp::executor::AnyExecutable>) () from /home/firesurfer/workspace/ros2_ws/install/lib/librclcpp.so
#5  0x00007ffff1fcbdf9 in rclcpp::executor::Executor::spin_once(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/firesurfer/workspace/ros2_ws/install/lib/librclcpp.so
#6  0x000000000069d5a1 in rclcpp::executor::Executor::spin_until_future_complete<std::shared_ptr<ros2_components_msg::srv::ListChilds_Response_<std::allocator<void> > >, std::ratio<1l, 1000l> > (
    this=0x7fffffffd3f0, future=..., timeout=...) at /home/firesurfer/workspace/ros2_ws/install/include/rclcpp/executor.hpp:184
#7  0x000000000068c94d in rclcpp::executors::spin_node_until_future_complete<std::shared_ptr<ros2_components_msg::srv::ListChilds_Response_<std::allocator<void> > >, std::ratio<1l, 1000l> > (executor=..., 
    node_ptr=std::shared_ptr (count 12, weak 1) 0x974440, future=..., timeout=...) at /home/firesurfer/workspace/ros2_ws/install/include/rclcpp/executors.hpp:67
#8  0x0000000000679fe9 in rclcpp::spin_until_future_complete<std::shared_ptr<ros2_components_msg::srv::ListChilds_Response_<std::allocator<void> > >, std::ratio<1l, 1000l> > (
    node_ptr=std::shared_ptr (count 12, weak 1) 0x974440, future=..., timeout=...) at /home/firesurfer/workspace/ros2_ws/install/include/rclcpp/executors.hpp:81

```
gdb output:
```
Program received signal SIGSEGV, Segmentation fault.
0x00000000006747b8 in std::__shared_ptr<std::__future_base::_State_baseV2, (__gnu_cxx::_Lock_policy)2>::operator-> (this=0x0) at /usr/include/c++/5/bits/shared_ptr_base.h:1055
1055		return _M_ptr;

```

jacquelinekay, 2016-01-31:
Thanks for reporting this. Can you post a minimal working example that compiles online somewhere and link to it here (perhaps via Gist)?

firesurfer, 2016-02-01:
Hey, unfortunately I don't have the time to create a minimal example at the moment. I try to give one at the end of the week. 

At the moment I did a small fix in  rclcpp/client.hpp:

in the function handle_response
```c++
if(call_promise)
{
    call_promise->set_value(typed_response);
    callback(future);
}
```

With this fix everthing is working without any segfaults.
But I don't think that this should be permanent fix.

firesurfer, 2016-02-03:
Hey,
I create a minimum demo on:
https://github.com/firesurfer/Segfault_demo

It consists of two parts: a segfault demo server and a segfault demo client. The segfault occurs in the client (test.cpp)

jacquelinekay, 2016-02-03:
Thanks. I'll take a look at this today.

jacquelinekay, 2016-02-03:
The segfault is generalizable to services/clients, it's not specific to parameters (parameters are implemented on top of services/clients). I rewrote the test with a plain service and client and reproduced the segfault in ros2/system_tests#104.

My potential fix is in ros2/rclcpp#193. It's not much more complicated than what you did, but I'm also not sure if that should be the permanent fix.

dirk-thomas, 2016-02-12:
Based on @gerkey investigation I think we should first merge these two (ros2/rmw_connext#131 ros2/rmw_opensplice#112) once the Windows CI job turns over and is green. They have the potential to fix other issues (like 100% cpu load, etc.) too.

gerkey, 2016-02-12:
After some investigation, with consultation from @jacquelinekay and @dirk-thomas, I believe the following things:

* The proposed fix in this PR addresses the client-side segfault. I haven't seen any client-side problems at all.
* With the client-side problem fixed, this test case now exposes a server-side problem with services.
* The timeout seen be Jenkins is due to the service server getting into a bad state in between the two service calls being made by the client.
* The effect of the bad state is that `wait()` continuously returns the service-associated read condition as something that should be checked, but `rmw_take_request()` continuously fails to take any request data; this loop proceeds forever, causing `spin_node_some()` to never return.
* The bad state is caused by something happening to the read condition associated with the service server when the first service client disconnects. I don't know where the bad state originates; it might be in OpenSplice, or in rmw_opensplice_cpp, or nowhere, in the sense that we should be handling an event in rmw_opensplice_cpp that we're currently not.
* A related issue is that we're doing unnecessary work because we're not nulling service (and other) handles when `wait()` times out; @dirk-thomas addressed that in ros2/rmw_opensplice#112.
* All of the above might also apply to other rmw implementations (e.g., ros2/rmw_connext#131).

jacquelinekay, 2016-02-12:
I'm still seeing a segfault on master with the Opensplice version of the test in ros2/system_tests#104, and an occasional deadlock with ros2/rclcpp#193.

gerkey, 2016-02-12:
@jacquelinekay That's what I'd expect: running ros2/system_tests#104 without ros2/rclcpp#193 gives you a segfault (essentially the original bug reported here) and running it with ros2/rclcpp#193 gives you a deadlock (caused by the "bad state" that the service server gets into upon client disconnection). @dirk-thomas's fix in ros2/rmw_opensplice#112 addresses a similar issue but does not get rid of the deadlock.

jacquelinekay, 2016-02-12:
Got it, I thought you were suggesting those PRs would fix the issue, but you're saying that the failure might be due to related issues.

gerkey, 2016-02-12:
Yeah, there's definitely something still lurking in the management of service server read conditions when a client disconnects. We were able to demonstrate the problem last night with a single client that calls the service, then sleeps for several seconds, then exits. The service server looks good at first, responds correctly to the service call, still looks good, then goes bad immediately when the client exits. I'm hoping to look further into it today.

dirk-thomas, 2016-02-12:
While this line listens to any instances:

https://github.com/ros2/rmw_opensplice/blob/5a56edf889f563d52be002258e6f4722cf9ddda2/rmw_opensplice_cpp/src/rmw_service.cpp#L120

that line only takes the alive ones:

https://github.com/ros2/rmw_opensplice/blob/5a56edf889f563d52be002258e6f4722cf9ddda2/rosidl_typesupport_opensplice_cpp/resource/srv__type_support.cpp.template#L91

The result is that once a sample with non-alive instance arrives opensplice wakes up from rmw_wait immediately forever resulting in a busy loop. That loop happens within a single `spin_*` call within rclcpp.

gerkey, 2016-02-12:
So changing to ANY_INSTANCE_STATE in the take() call will fix it?

dirk-thomas, 2016-02-12:
Hopefully yes, running CI jobs for it right now...

dirk-thomas, 2016-02-12:
New CI jobs:

* http://ci.ros2.org/job/ci_linux/976/
* http://ci.ros2.org/job/ci_osx/811/

dirk-thomas, 2016-02-12:
Creating a second client after the first one was destroyed fails with Connext dynamic. Looking into it...

dirk-thomas, 2016-02-13:
New CI jobs (including ros2/rmw_connext#134):

* http://ci.ros2.org/job/ci_linux/984/
* http://ci.ros2.org/job/ci_osx/816/
* http://ci.ros2.org/job/ci_windows/1067/

tfoote, 2016-02-16:
From the meeting. Merge this as is with the return after a warning. Add a todo, and after the alpha reescalate it to an exception so we can find it if it happens in the future.

closed 2016-02-16 by dirk-thomas