ros2/rclcpp #195: Segfault when using Parameter Client with spelling like DemoRobot

firesurfer, 2016-02-16:
When trying to create a demo application for #193 I got another segfault when using a Parameter Client with a spelling like DemoRobot for the parameter name. 

Have a look at my demo application https://github.com/firesurfer/ros2_components_demo
It need the qt5 dev package in order to compile.

The application consists of three parts.
demo_parameter_server
demo_server
demo_client

The segfault happens in the demo_client in the file /src/ros2_components/Robot.h in the function ListAllKnownRobots

jacquelinekay, 2016-02-16:
To clarify, you only see the segfault if the name of the parameter is "DemoRobot"? Are you suggesting that the CamelCase name is the cause of the segfault? Or the particular string? That seems to be a dubious cause for the issue. Can you make a more minimal example that reproduces the segfault?

A debugger backtrace from the segfault would also be helpful, if you don't have time to make a MWE.

dirk-thomas, 2016-02-17:
I tried to compile your demo application which took quite a while and some updates. Please make sure that your examples compile correctly out of the box against the latest default branches in the future.

That being said I can reproduce the segfault in the demo_client now:

```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7b053f7 in rclcpp::parameter::ParameterVariant::get_value<(rclcpp::parameter::ParameterType)1> (this=0x0)
    at /home/dthomas/ros/ros2/ws_ros2/src/ros2/rclcpp/rclcpp/include/rclcpp/parameter.hpp:120
120	    if (value_.type != rcl_interfaces::msg::ParameterType::PARAMETER_BOOL) {
(gdb) bt
#0  0x00007ffff7b053f7 in rclcpp::parameter::ParameterVariant::get_value<(rclcpp::parameter::ParameterType)1> (this=0x0)
    at /home/dthomas/ros/ros2/ws_ros2/src/ros2/rclcpp/rclcpp/include/rclcpp/parameter.hpp:120
#1  0x00007ffff7b03a06 in rclcpp::parameter::ParameterVariant::as_bool (this=0x0) at /home/dthomas/ros/ros2/ws_ros2/src/ros2/rclcpp/rclcpp/src/rclcpp/parameter.cpp:148
#2  0x00000000004188c7 in ros2_components::Robot::ListKnownRobots (prefix=..., _parentNode=...) at /home/dthomas/ros/ros2/ws_demog/src/ros2_components_demo/src/ros2_components/Robot.h:157
#3  0x000000000041782b in main (argc=1, argv=0x7fffffffb0a8) at /home/dthomas/ros/ros2/ws_demog/src/ros2_components_demo/src/exec/client.cpp:11
```

dirk-thomas, 2016-02-17:
The client segfaults due to two bugs in your code:

* You must check the return value of `spin_until_future_complete` in https://github.com/firesurfer/ros2_components_demo/blob/fc7d46547d93477ac1f2a0c61a3880e5d810d757/src/ros2_components/Robot.h#L156 to ensure it is `rclcpp::executor::FutureReturnCode::SUCCESS` - lucky enough it is so this was not causing the segfault
* You must check the size of vector retrieved from the future in https://github.com/firesurfer/ros2_components_demo/blob/fc7d46547d93477ac1f2a0c61a3880e5d810d757/src/ros2_components/Robot.h#L157 before accessing the first element - and this one bit you since the vector size is zero in this case

closed 2016-02-17 by dirk-thomas