ros2/rclcpp #196: Feature / Question - List all parameters of a parameter server

firesurfer, 2016-02-17:
Is it possible to list all parameters of a parameter server ?
I created a workaround in https://github.com/firesurfer/ros2_components/blob/master/src/ros2_components/Robot.h#L119 by requesting about 20.000 possible parameter names. It would be nicer if it would be possible to get a list of all parameters

dirk-thomas, 2016-02-17:
The list parameters service call (https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/srv/ListParameters.srv) should provide that functionality. The service call is exposed as API in rclcpp as https://github.com/ros2/rclcpp/blob/8c47f6151bd5c68bd7b2a31e375c114ad8ba087e/rclcpp/include/rclcpp/parameter_client.hpp#L146-L150

firesurfer, 2016-02-17:
Yeah. But what to pass for as prefix if I want to list all parameters? I already tried a empty list or * but it didn't work as far as I remember.

dirk-thomas, 2016-02-17:
From looking at the implementation (https://github.com/ros2/rclcpp/blob/69f7bca85d523ee8647c2051651184023ad1c6ad/rclcpp/src/rclcpp/node.cpp#L236) I can't figure out how this is currently possible. Maybe @esteve can comment on this?

It would be good to also document the necessary arguments of this common use case in the .srv file.

gerkey, 2016-02-18:
ros2/rclcpp#197 adds the ability to list all parameters by supplying an empty prefix list. ros2/system_tests#107 adds a test for it.

dirk-thomas, 2016-02-18:
@gerkey Can you please run a CI job for it.

gerkey, 2016-02-18:
I'd like to get feedback on the approach. Is an empty prefix list and ignored depth a good way to list all parameters?

dirk-thomas, 2016-02-18:
A think returning all parameters when an empty prefix list is passes is good.

But the depth parameter should still be honored in that case.

gerkey, 2016-02-19:
Proposal: when the prefix list is empty, a depth of `D` means: return all parameters whose names contain `<=D` separators (currently the `.` character). Does that make sense?

dirk-thomas, 2016-02-19:
I don't think `<=D` makes sense. `D=1` should return everything *without* a separator - basically the first level. And of course also the parent group name for everything with a separator.

And I assume `depth=0` has a similar semantic as an empty prefix list. So return all parameters with an arbitrary number of separators.

firesurfer, 2016-02-19:
This doens't directly affects the approaches discussed before.
From the point of usability I think it's more convenient to have two methods in the rclcpp parameter_client.hpp.
The same as at the moment
```
list_parameters(
    const std::vector<std::string> & parameter_prefixes,
    uint64_t depth);
```
and another variant without any arguments
```
list_parameters();
```
I think that would be less confusing than empty arguments. Internally the second method could just call the first one with an empty parameter_prefixes vector.

gerkey, 2016-02-19:
I like the idea of adding the second signature of `list_parameters()` with no args. But we still need to decide the semantics for this query when it happens via a service call, independent of any particular client library. That is, we need to decide how to interpret the contents of the request in the [rcl_interfaces/ListParameters service](https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/srv/ListParameters.srv).

An empty `prefixes` list makes sense as a request for all parameters. And now that I've looked at the service spec, which defines `DEPTH_RECURSIVE=0` (I didn't know that previously), I agree with @dirk-thomas's suggestion of having `D==0` mean all depths, and `D>0` mean all params with `<D` separators. I'll do that.

tfoote, 2016-02-19:
I agree that an empty list could be a special case, but I'd also argue that if you pass a zero length string as a prefix that should also match everything. Otherwise if you're passing an argument through from another source you would have to add special logic to detect the zero length prefix and follow a special special logic if you want to support the zero length prefix.

dirk-thomas, 2016-02-19:
I don't think a prefix list with one empty string had the same semantic. I understood the semantic of the prefix as follows: a prefix `foo`

* matches all parameters grouped under `foo` which means start with `foo.`.
* it does **not** match e.g. parameter `foobar`

tfoote, 2016-02-19:
The question is whether you consider there to be an implicit separator at
the beginning.

foo matches foo.* but not foobar could be rewritten as
prefix matches ."prefix".*

but if you consider it more like a path with a leading separator and empty
string matches .* implying both foo and foobar would match.

If '' does not collapse and match any, I would expect that then we could
have zero length namespaces.
With that you would get things like `..foo..bar` as distinct from .foo.bar.

​

gerkey, 2016-06-20:
In #230, I implemented the following strategy:

* An empty `prefixes` list means that any parameter name will match.
* Given the matches against `prefixes`, the `depth` argument is used to further filter the result to be those parameters that contain `< depth` separators. I.e., `depth==1` means the first level, no separators allowed; `depth==2` means the first and second level, at most 1 separator allowed; and so on. As a special case, `depth==0` means all depths.

In https://github.com/ros2/system_tests/pull/140 I added new tests for the empty prefixes case and updated other cases to account for the new handling of the `depth` argument.

gerkey, 2016-06-20:
* Linux:
 * [image]
* OSX:
 * [image]
* Windows:
 * [image]

gerkey, 2016-06-21:
New feature (belatedly) merged. Thanks for the request!

closed 2016-06-21 by gerkey