ros2/rclcpp #200: parameter API does not allow to distinguish error cases

dirk-thomas, 2016-03-01:
E.g. https://github.com/ros2/rclcpp/blob/249b7d80d8f677edcda05052f598de84f4c2181c/rclcpp/src/rclcpp/parameter_client.cpp#L246-L257 return an empty vector if something goes wrong which is also a valid response.

jacquelinekay, 2016-03-09:
Since one of my goals for this alpha is "C Parameters" (whatever that means--since one of the questions I need to address is how much of parameter functionality should be in rcl vs. rclcpp anyway) I will also think about improving the parameter API.

jacquelinekay, 2016-03-09:
You could also argue that an exception could be thrown in rclcpp instead of, say, changing the API to return an error code (which is what I'm prototyping right now, and what would have to happen in rcl).

wjwwood, 2020-03-19:
Updated link:

https://github.com/ros2/rclcpp/blob/96ebf59a6045a535730d98fff25e522807c7aa75/rclcpp/src/rclcpp/parameter_client.cpp#L412-L426