ros2/rclcpp #214: Strg+c doesn't exit application anymore

firesurfer, 2016-04-28:
After the rclcpp was switched to use rcl I can't use Strg+c anymore to exit my application.

This is the code I use:

```c++
#include <iostream>
#include "rclcpp/rclcpp.hpp"
#include <memory>

rclcpp::node::Node::SharedPtr node;
void spin()
{
    rclcpp::executors::SingleThreadedExecutor exec;
    std::cout << "spinning" << std::endl;
    rclcpp::WallRate loop_rate(20);
    while(rclcpp::ok())
    {
        exec.spin_node_some(node);
        loop_rate.sleep();
    }
}
int main(int argc, char **argv)
{
    rclcpp::init(argc, argv);

    node = rclcpp::node::Node::make_shared("DemoClient");
    std::thread spinner(&spin);
    rclcpp::WallRate loop_rate(30);
    std::string input = "";
    while(input != "exit")
    {   std::getline (std::cin,input);

        loop_rate.sleep();
    }

    return 0;
}

```

By typing "exit" the application will stop but prints `terminate called without an active exception`

jacquelinekay, 2016-04-28:
That's my fault... I'll take a look today.

jacquelinekay, 2016-05-03:
I can confirm that the exit signal doesn't work for parameter clients. If I press ctrl+c during "spin_until_future_complete", the thread 

Regarding @firesurfer's example: I can make two modifications to the example to make it work as expected:

- Remove the getline logic and simply use `std::thread::join` (block until the other thread is finished).

```
    node = rclcpp::node::Node::make_shared("DemoClient");
    std::thread spinner(&spin);
    spinner.join();
```

- Use `rclcpp::ok()` in the while loop and use `thread::join` to make sure the other thread is cleaned up properly..
```
    node = rclcpp::node::Node::make_shared("DemoClient");
    std::thread spinner(&spin);
    rclcpp::WallRate loop_rate(30);
    std::string input = "";
    while(input != "exit" && rclcpp::ok())
    {   std::getline (std::cin,input);

        loop_rate.sleep();
    }
    spinner.join();
    return 0;
```

I believe the "getline" logic creates a problem where the main thread will only exit until it gets the expected "exit" token. Therefore CTRL-C works to stop the "spinner" thread but not the main thread.

Additionally, CTRL/STRG-C still works with the `talker` and `listener` examples. Therefore I think the problem is with `spin_until_future_complete`, which is used for parameter clients, where you probably originally observed the problem.

jacquelinekay, 2016-05-04:
So, I was mistaken thinking that the issue was originally with spin_until_future_complete, since CTRL+C works with clients, and it even works with this remote parameter client example:

https://github.com/ros2/examples/blob/remote_parameters/rclcpp_examples/src/parameters/remote_parameter_client.cpp

I realized that the example I was originally using to test this had an issue that would would cause it to block indefinitely on CTRL+C, which is that it tries to access a `std::future` that isn't ready yet.

Perhaps there should be a solution where, when sigint is called, unfulfilled `std::promise` objects stored by clients should receive an exception value, which seems to be the way to indicate an error state in promises in the std library.

@firesurfer, did the example you gave work before the rcl/rclcpp refactor (e.g. does it work in alpha5)? And does STRG+C work with very basic examples (e.g. talker/listener)?

firesurfer, 2016-05-04:
1. Your 2nd propsed fix actually works `while(input != "exit" && rclcpp::ok())`

2. My example worked before the refactoring

3. STRG+C works with talker and listener

dirk-thomas, 2017-01-03:
@firesurfer Can you please recheck if this is still a problem with the beta 1.

firesurfer, 2017-06-05:
@dirk-thomas I didn't recieve any notification about your last comment.
I can confirm that the issue doesn't exist anymore. The `rclcpp::ok()` returns false in case STRG+C was called.  (The code I'm currently using: https://github.com/firesurfer/ros2_components/blob/master/src/ros2_components/ManagedNode.cpp#L92)

closed 2017-06-05 by firesurfer