ros2/rclcpp #216: Massive memory leak in AsyncParameterClient

firesurfer, 2016-05-02:
I got a massive memory leak (about 30 GB of RAM in 3 sec) due to the AsyncParameterClient

Some demo code:
```c++
rclcpp::node::Node::SharedPtr node;
void spin()
{
    rclcpp::executors::SingleThreadedExecutor exec;
    std::cout << "spinning" << std::endl;
    rclcpp::WallRate loop_rate(20);
    while(rclcpp::ok())
    {
        exec.spin_node_some(node);
        loop_rate.sleep();
    }
}
int main(int argc, char *argv[])
{
    rclcpp::init(argc, argv);

    node = rclcpp::node::Node::make_shared("Leak test");
    std::thread spinner(&spin);

    std::vector<rclcpp::parameter_client::AsyncParametersClient::SharedPtr> holder2;
    for(int i = 0; i < 100;i++)
    {
        rclcpp::parameter_client::AsyncParametersClient::SharedPtr test2 =std::make_shared<rclcpp::parameter_client::AsyncParametersClient>(node, "ParameterServer");
        holder2.push_back(test2);

    }

    rclcpp::WallRate loop_rate(30);
    std::string input = "";
    while(input != "exit")
    {   std::getline (std::cin,input);

        loop_rate.sleep();
    }

}

```

dirk-thomas, 2016-05-02:
Can you please provide more information on the rmw implementation you are using as well as which alpha version of ROS 2. Also the platform and build type might be relevant.

jacquelinekay, 2016-05-02:
possibly related (if you are using Opensplice): ros2/rmw_opensplice#128

firesurfer, 2016-05-02:
@dirk-thomas 

I checked out the master branch for all packages and I'm using opensplice. The platform I'm using is an up to date debian testing with gcc 5.3.1.

@jacquelinekay  I don't think it's related to https://github.com/ros2/rmw_opensplice/issues/128 

1. This bug occured after rclcpp was switched to rcl (I used the ParameterClient before and it didn't took that much memory)
2. I tried creating publishers instead of AsyncParameterClients. In this case the application didn't took a significant amount of memory.

jacquelinekay, 2016-05-02:
ok. I'll take a look with valgrind.

dirk-thomas, 2016-05-02:
I see the same / similar problem with FastRTPS (couldn't reproduce it with OpenSplice yet). Easy to reproduce command: `/usr/bin/time -v list_parameters__rmw_fastrtps_cpp`

firesurfer, 2016-05-03:
Could it be possible that due to some error in the build system also the `....__rmw_opensplice_cpp` is build agaist FastRTPS ? (Or I did something wrong in my CMake-File)
Because even though I explicitly used the generated `....__rmw_opensplice_cpp` executable valgrind tells me that there are bytes lost in FastRTPS ?

```
==7499== Syscall param sendmsg(msg.msg_iov[0]) points to uninitialised byte(s)
==7499==    at 0x0a6a444d: ??? (syscall-template.S:84)
==7499==    by 0x0f2182e1: boost::asio::detail::socket_ops::sendto(int, iovec const*, unsigned long, int, sockaddr const*, unsigned long, boost::system::error_code&) (socket_ops.ipp:1299)
==7499==    by 0x0f218388: boost::asio::detail::socket_ops::sync_sendto(int, unsigned char, iovec const*, unsigned long, int, sockaddr const*, unsigned long, boost::system::error_code&) (socket_ops.ipp:1321)
==7499==    by 0x0f219aaa: unsigned
==7499==    by 0x0f219480: unsigned
==7499==    Address 0x65c7d9b2 is 82 bytes inside a block of size 28,797 alloc'd
==7499==    at 0x04c29c0f: malloc (vg_replace_malloc.c:299)
==7499==    by 0x0f22331d: eprosima::fastrtps::rtps::CDRMessage_t::CDRMessage_t(unsigned int) (CDRMessage_t.h:71)
==7499==    by 0x0f223449: eprosima::fastrtps::rtps::RTPSMessageGroup_t::RTPSMessageGroup_t(unsigned int) (RTPSMessageGroup.h:42)
==7499==    by 0x0f222b21: eprosima::fastrtps::rtps::RTPSWriter::RTPSWriter(eprosima::fastrtps::rtps::RTPSParticipantImpl*, eprosima::fastrtps::rtps::GUID_t&, eprosima::fastrtps::rtps::WriterAttributes&, eprosima::fastrtps::rtps::WriterHistory*, eprosima::fastrtps::rtps::WriterListener*) (RTPSWriter.cpp:33)
==7499==    by 0x0f22376c: eprosima::fastrtps::rtps::StatefulWriter::StatefulWriter(eprosima::fastrtps::rtps::RTPSParticipantImpl*, eprosima::fastrtps::rtps::GUID_t&, eprosima::fastrtps::rtps::WriterAttributes&, eprosima::fastrtps::rtps::WriterHistory*, eprosima::fastrtps::rtps::WriterListener*) (StatefulWriter.cpp:64)
==7499==    Uninitialised value was created by a heap allocation
==7499==    at 0x04c29c0f: malloc (vg_replace_malloc.c:299)
==7499==    by 0x0a47830a: rmw_fastrtps_cpp::TypeSupport::createData() (TypeSupport.cpp:677)
==7499==    by 0x0a46901d: rmw_publish (functions.cpp:488)
==7499==    by 0x0aaba10f: rcl_publish (publisher.c:130)
==7499==    by 0x005a65a3: rclcpp::publisher::Publisher<ros2_components_msg::msg::EntityAdvertisement_<std::allocator<void>
==7499==

```

dirk-thomas, 2016-05-03:
In your CMake file you are linking against `rclcpp`. But `rclcpp` already links against one specific rmw implementation - on this case FastRTPS. You need to link against `rclcpp${target_suffix}` instead (same as e.g. here https://github.com/ros2/examples/blob/master/rclcpp_examples/CMakeLists.txt ).

dirk-thomas, 2016-05-31:
I just reran `/usr/bin/time -v list_parameters__rmw_fastrtps_cpp` with the latest version of FastRTPS and the massive memory usage seems to be not the case anymore. @firesurfer Can you please recheck and comment if the problem still exists for you with a current build.

dirk-thomas, 2016-06-14:
I will close this for now as it seems that this has been fixed. Please feel free to comment and if the problem still persists this can be reopened.

closed 2016-06-14 by dirk-thomas