ros2/rclcpp #217: Using OpenSplice: Could not initialize rcl node: error not set

firesurfer, 2016-05-03:
I fixed my wrong cmake file so the executable links to opensplice (See #216 )
Now I get the following error:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not initialize rcl node: error not set
```

This also happens if I try to run the `talker__rmw_opensplice_cpp` example

jacquelinekay, 2016-05-03:
I believe executables linked to Opensplice require the `install/setup.bash` file to be sourced before running.

firesurfer, 2016-05-03:
Yeah that's the case. Thanks!

dirk-thomas, 2016-05-03:
I got exactly the same error message but I don't agree with closing this. The error message is completely useless. The message should state `OSPL_URI not set` but the code which intends to set the error message doesn't work correctly. As far as I can see the following line is broken: https://github.com/ros2/rcl/blob/a01943973a29ca3699a7c026a5a0a0b4acf01e18/rcl/src/rcl/node.c#L102-L103

dirk-thomas, 2016-05-06:
Closed with ros2/rmw#68

closed 2016-05-06 by dirk-thomas