ros2/rclcpp #218: Question: Possible to determine running nodes in a system?

firesurfer, 2016-05-06:
I have the task/problem to find a specific node / publisher that runs on a computer I only know the ip address from. Is this at the moment possible with ROS 2 or do I have to create my own socket that listens on a specifc port on this computer and provides the information I need to know?

wjwwood, 2016-05-06:
I'm not sure I understand the question. If you want to list all of the nodes in the ROS graph then:

Not yet, we're still in the process of writing the "graph" related API, which allows you to ask questions about the graph, like what nodes exist.

If you want to list the publishers in the ROS graph then you can use `rmw_get_topic_names_and_types`:

https://github.com/ros2/rmw/blob/master/rmw/include/rmw/rmw.h#L241-L252

If you mean "given an IP what is the node on that IP", then that doesn't really make sense, because there can be multiple Nodes per IP. If the question is "given an IP what are all the nodes on that IP", then I expect you'll be able to answer that once we've implemented the node part of our graph API, as it would probably include the machine and process and listening IP(s) for each node.

wjwwood, 2016-05-06:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2016-05-06 by wjwwood