ros2/rclcpp #219: Is there a maximum message size / maximum amount of fields in a message?

firesurfer, 2016-05-07:
I got a rather large message in terms of many fields in a message:

```
int64 id

string componentname

string type

int64 parent

string parenttype

builtin_interfaces/Time stamp

int32 advertisementtype

int64[] childids

string[] childtypes

int64 machineip
```

I recognized while testing with opensplice as middleware that in some transfered messages the last fields are empty. For example machineip is in most cases 0. The field at the beginning of the message are transfered correctly. 

Is this a bug or specified behaviour ?

dirk-thomas, 2016-05-07:
Ten fields doesn't look large to me. We already have messages with more fields in use.

And none of the fields is large in size. All strings and arrays are dynamic - no large fixed size arrays. So I don't think the message size is large either. Above 64k for a serialized message the datagrams needs to be fragmented but that didn't seem to be the case here.

firesurfer, 2016-05-08:
Okey, then I have to report another bug. 

Setup: 
1 Process with 1 publisher (with the message definition above) and 1 subscription. This subscription recieves the complete message. (Publisher and subscription running in the same node)

1 Process with subscription to the same topic (Subscription running in an different node). Here I got about half of the message cut off in about 80% of the cases. By cut off I mean the last few fields are empty. If I move the last field (for example machineip) to the top it will be recieved correct in 100% of the cases. 

I' trying to create a demonstration application of this bug the next days.

firesurfer, 2016-05-09:
I could reproduce the bug in simple demo. Please have look at https://github.com/firesurfer/message_bug

It consists of two executables:

message_bug_exec and message_bug_remote_exec

`message_bug_exec` publishes messages and also subscribes them. `message_bug_remote_exec` simply subscribes them. You will notice that in `message_bug_exec` the last field is recieved correct whereas in `message_bug_remote_exec` the last field is 0 even though it should be 100.

dirk-thomas, 2016-05-09:
I can confirm the problem with OpenSplice. With Connext and FastRTPS the received message seems to be correct. Also when using a non OpenSplice publisher the OpenSplice subscriber shows the correct result. So it looks like a bug in the publishing implementation of `rmw_opensplice_cpp`.

dirk-thomas, 2016-05-09:
I updated our existing communication tests to cover this case: ros2/system_tests#118

dirk-thomas, 2016-05-13:
This seems to be a bug in OpenSplice itself: PrismTech/opensplice#25

jacquelinekay, 2016-06-06:
The re-enabled test passed on Linux:

http://ci.ros2.org/job/ci_linux/1393/testReport/%28root%29/test_publisher_subscriber_cpp__dynamicarrayprimitives__rmw_opensplice_cpp__rmw_fastrtps_cpp_/test_publisher_subscriber/

I will run another job with all publisher_subscriber tests to see how it does.

jacquelinekay, 2016-06-06:
I see, Opensplice to FastRTPS succeeds but Opensplice to itself does not. Looks like the issue did not magically fix itself.

dirk-thomas, 2017-08-30:
Seems to work with OpenSplice 6.7, see ros2/system_tests#222.

closed 2017-08-30 by dirk-thomas