ros2/rclcpp #232: FastRTPS doesn't send message if publisher and subscribtion don't have the same qos_profile

firesurfer, 2016-06-30:
I switched to fastRTPS as middleware because of bug #219 . I realized that messages won't be transmitted if a publisher uses the rmw_qos_profile_sensor_data qos profile and the subscription uses the rmw_qos_profile_default  qos profile. Is this intended behaviour?

wjwwood, 2016-06-30:
If the publisher is unreliable and the subscriber is reliable, which I believe is the case in your scenario, then yes I would not expect them to communicate. This makes sense because the publisher is unable to delivery messages as the subscribe would like.

The opposite is not true however. A reliable publisher can publish to an unreliable subscriber.

firesurfer, 2016-07-01:
Okey, I just want to note that opensplice transmitted the message in this configuration.
I close this issue because it's apparently intended behaviour.

closed 2016-07-01 by firesurfer

wjwwood, 2016-07-01:
We recently fixed a bug which was causing the default qos to be used always even if you specified a different one. I wonder if that was the issue. I would have expected opensplice to not communicate in that configuration either.