ros2/rclcpp #238: ParameterVariant doesn't allow setting of name or value

gerkey, 2016-07-08:
The `ParameterVariant` class requires the name and value (and thus type) of the variable to be passed to the constructor. We should add `set_name()` and `set_value()` methods.

gerkey, 2016-07-09:
Related enough to be mentioned here: `ParameterVariant` should allow setting from a `rcl_interfaces::msg::ParameterValue`.

sloretz, 2018-07-05:
I changed the title since `ParameterVariant` was split into `Parameter` and `ParameterValue` in #481 and `Parameter` still doesn't have `set_name()` and `set_value()` methods.

dirk-thomas, 2018-07-05:
We are using member-based access in ROS message classes. So I don't think we should add a method for these two fields.

In C++ we generally don't have constructors with arguments (since the order would be problematic in case of future changes to the message). So the user has to set all fields after creating an instance.

So I am not sure what the scope of this ticket is. If it is just about adding methods to set `name` and `value` (over setting them as members) I would suggest to close it.

dirk-thomas, 2018-07-05:
It wasn't clear from the description that this ticket is referring to the `rclcpp` class. I was assuming it was referring to the `rcl_interfaces` with that name. For the C++ class it would make sense to add setters.

We can either add those or just close this ticket since this is more a nice to have rather than preventing the user to use the class.