ros2/rclcpp #240: FastRTPS -  free(): corrupted unsorted chunks

firesurfer, 2016-07-11:
I got this very simple code:

https://github.com/firesurfer/ros2_components/blob/master/src/tests/create_robot.cpp

I keep having an error like this: (Not the complete output, the complete output is very long)
```
*** Error in `./ros2_components_create_robot_test': free(): corrupted unsorted chunks: 0x000000000103c9d0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x71fc5)[0x7f45af70cfc5]
/lib/x86_64-linux-gnu/libc.so.6(+0x77966)[0x7f45af712966]
/lib/x86_64-linux-gnu/libc.so.6(+0x7814e)[0x7f45af71314e]
/home/firesurfer/workspace/ros2_ws/install/lib/libfastrtps.so(_ZN8eprosima8fastrtps4rtps14CDRMessagePoolD1Ev+0x5a)[0x7f45aebda1ba]
/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0x8f)[0x7f45af6d114f]
/home/firesurfer/workspace/ros2_ws/install/lib/libfastrtps.so(+0x71c33)[0x7f45aeb99c33]
======= Memory map: ========
00400000-00413000 r-xp 00000000 103:02 7734542                           /home/firesurfer/workspace/iboss_ws/build/ros2_components/src/tests/ros2_components_create_robot_test
00612000-00613000 rw-p 00012000 103:02 7734542                           /home/firesurfer/workspace/iboss_ws/build/ros2_components/src/tests/ros2_components_create_robot_test
00dce000-01b90000 rw-p 00000000 00:00 0                                  [heap]

```

I could track down the error to the CMakeFile ( https://github.com/firesurfer/ros2_components/blob/master/src/tests/CMakeLists.txt ):
When I link to my ros2_components lib, even if I don't call any functions from it I keep getting the error from above. If I remove the link statement the application exits correct.

Could you give me some advice regarding this error ?

dirk-thomas, 2016-07-12:
I was unable to compile the code to check if the problem happens for me as well. I clone the repos `ros2_components`, `ros2_components_msg`, and `ros2_simple_logger` (in the future please provide a rosinstall file (or at least a list) with all the required repositories to make it as easy as possible to reproduce a problem).

The `ros2_simple_logger` package is not using the include directories of `builtin_interfaces` which makes it failed when built in isolation. I was able to fix that easily. But the executables in  `ros2_simple_logger` failed to link due to undefined references. Is that not happening for you? What version of ROS 2 are you using? On which platform? With which rmw implemenations?

firesurfer, 2016-07-12:
I could fix the bug myself. I had bit of a mess with a singleton and a shared_ptr. 
Sry for bothering you with that. I wasn't sure because of the error output which stated something about fastrtps and the debugger always jumped into the destruction of a ros2 publisher.

The `ros2_simple_logger` actually compiles fine for me. I'm using debian stretch, gcc 5.4.0 and I'm using the master branches of all ROS 2 repositories. (Doing a pull and a rebuild every few days). I'm currently using `rmw_fastrtps_cpp` .  As you can see in the cmake file there aren't special dependencies. Just a bit messing around because I had trouble using messages which are located and build in the same package as the code which uses the messages. 

Sry about not providing a repository list.

firesurfer, 2016-07-13:
It seems like I unfortunately need to reopen this issue, because I absolutely got no clue how to track down this error. I would really appreciate if you could have a look at it, even though I'm not sure the bug isn't in my own code. As said before I get an ` corrupted unsorted chunks` error when the application exits. 

For reproduction of the error  it should be enough to run the `ros2_simple_logger_test` in my https://github.com/firesurfer/ros2_simple_logger repository. The current commit on the master branch compiles fine on my system even in an isolated workspace. On my system (see the previous post for information about my setup) I get the following error (this time the complete output):

```
firesurfer@firehome:~/workspace/iboss_ws/install/tests$ ./ros2_simple_logger_test 
GCC Version:5
spinning
2016-07-13 17:28:58 : Debug :  debug stream test
2016-07-13 17:28:58 : Info  :  info stream test
2016-07-13 17:28:58 : Important  :  important stream test
2016-07-13 17:28:58 : Warning  :  warning stream test
2016-07-13 17:28:58 : Error  : exception stream test
2016-07-13 17:28:58 : Fatal  : fatal stream test
2016-07-13 17:28:58 : Warning  : this should show up
2016-07-13 17:28:58 : Important  :  important stream test
2016-07-13 17:28:58 : Warning  :  warning stream test
2016-07-13 17:28:58 : Error  : exception stream test
2016-07-13 17:28:58 : Fatal  : fatal stream test
2016-07-13 17:28:58 : Warning  : this should show up
exit
2016-07-13 17:29:00 : Info  : exit
Going to exit now
stopped spinning node
*** Error in `./ros2_simple_logger_test': free(): corrupted unsorted chunks: 0x000000000271d410 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x6ef45)[0x7fce079eff45]
/lib/x86_64-linux-gnu/libc.so.6(+0x746b6)[0x7fce079f56b6]
/lib/x86_64-linux-gnu/libc.so.6(+0x74e9e)[0x7fce079f5e9e]
/home/firesurfer/workspace/ros2_ws/install/lib/libfastrtps.so(_ZN8eprosima8fastrtps4rtps14CDRMessagePoolD1Ev+0x5a)[0x7fce06ec01ba]
/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0x8f)[0x7fce079b6edf]
/home/firesurfer/workspace/ros2_ws/install/lib/libfastrtps.so(+0x71c33)[0x7fce06e7fc33]
======= Memory map: ========
00400000-00418000 r-xp 00000000 103:02 4196326                           /home/firesurfer/workspace/iboss_ws/build/ros2_simple_logger/ros2_simple_logger_test
00618000-00619000 rw-p 00018000 103:02 4196326                           /home/firesurfer/workspace/iboss_ws/build/ros2_simple_logger/ros2_simple_logger_test
024b4000-032c9000 rw-p 00000000 00:00 0                                  [heap]
7fcdd4000000-7fcdd4021000 rw-p 00000000 00:00 0 
7fcdd4021000-7fcdd8000000 ---p 00000000 00:00 0 
7fcddc000000-7fcddc021000 rw-p 00000000 00:00 0 
7fcddc021000-7fcde0000000 ---p 00000000 00:00 0 
7fcde0000000-7fcde0021000 rw-p 00000000 00:00 0 
7fcde0021000-7fcde4000000 ---p 00000000 00:00 0 
7fcde4000000-7fcde4021000 rw-p 00000000 00:00 0 
7fcde4021000-7fcde8000000 ---p 00000000 00:00 0 
7fcde97fb000-7fcde97fc000 ---p 00000000 00:00 0 
7fcde97fc000-7fcde9ffc000 rw-p 00000000 00:00 0 
7fcde9ffc000-7fcde9ffd000 ---p 00000000 00:00 0 
7fcde9ffd000-7fcdea7fd000 rw-p 00000000 00:00 0 
7fcdea7fd000-7fcdea7fe000 ---p 00000000 00:00 0 
7fcdea7fe000-7fcdeaffe000 rw-p 00000000 00:00 0 
7fcdeaffe000-7fcdeafff000 ---p 00000000 00:00 0 
7fcdeafff000-7fcdeb7ff000 rw-p 00000000 00:00 0 
7fcdeb7ff000-7fcdeb800000 ---p 00000000 00:00 0 
7fcdeb800000-7fcdec000000 rw-p 00000000 00:00 0 
7fcdec000000-7fcdec021000 rw-p 00000000 00:00 0 
7fcdec021000-7fcdf0000000 ---p 00000000 00:00 0 
7fcdf0000000-7fcdf0021000 rw-p 00000000 00:00 0 
7fcdf0021000-7fcdf4000000 ---p 00000000 00:00 0 
7fcdf4000000-7fcdf4021000 rw-p 00000000 00:00 0 
7fcdf4021000-7fcdf8000000 ---p 00000000 00:00 0 
7fcdf8000000-7fcdf8021000 rw-p 00000000 00:00 0 
7fcdf8021000-7fcdfc000000 ---p 00000000 00:00 0 
7fcdfc000000-7fcdfc021000 rw-p 00000000 00:00 0 
7fcdfc021000-7fce00000000 ---p 00000000 00:00 0 
7fce003f5000-7fce003f6000 ---p 00000000 00:00 0 
7fce003f6000-7fce00bf6000 rw-p 00000000 00:00 0 
7fce00bf6000-7fce00bf7000 ---p 00000000 00:00 0 
7fce00bf7000-7fce013f7000 rw-p 00000000 00:00 0 
7fce013f7000-7fce013f8000 ---p 00000000 00:00 0 
7fce013f8000-7fce01bf8000 rw-p 00000000 00:00 0 
[... 131 lines cut ...]
7fce0a8b8000-7fce0a9a6000 r-xp 00000000 103:02 13923714                  /usr/lib/libdcpssacpp.so
7fce0a9a6000-7fce0aba6000 ---p 000ee000 103:02 13923714                  /usr/lib/libdcpssacpp.so
7fce0aba6000-7fce0abd4000 r--p 000ee000 103:02 13923714                  /usr/lib/libdcpssacpp.so
7fce0abd4000-7fce0abd6000 rw-p 0011c000 103:02 13923714                  /usr/lib/libdcpssacpp.so
7fce0abd6000-7fce0abd7000 rw-p 00000000 00:00 0 
7fce0abd7000-7fce0abe3000 r-xp 00000000 103:02 13923707                  /usr/lib/libspliced.so
7fce0abe3000-7fce0ade3000 ---p 0000c000 103:02 13923707                  /usr/lib/libspliced.so
7fce0ade3000-7fce0ade4000 r--p 0000c000 103:02 13923707                  /usr/lib/libspliced.so
7fce0ade4000-7fce0ade5000 rw-p 0000d000 103:02 13923707                  /usr/lib/libspliced.so
7fce0ade5000-7fce0ae41000 r-xp 00000000 103:02 13923703                  /usr/lib/libdurability.so
7fce0ae41000-7fce0b040000 ---p 0005c000 103:02 13923703                  /usr/lib/libdurability.so
7fce0b040000-7fce0b041000 r--p 0005b000 103:02 13923703                  /usr/lib/libdurability.so
7fce0b041000-7fce0b042000 rw-p 0005c000 103:02 13923703                  /usr/lib/libdurability.so
[... cut ...]
7fce0b245000-7fce0b444000 ---p 00001000 103:02 13923699                  /usr/lib/libddsuser.so
7fce0b444000-7fce0b445000 r--p 00000000 103:02 13923699                  /usr/lib/libddsuser.so
7fce0b445000-7fce0b446000 rw-p 00001000 103:02 13923699                  /usr/lib/libddsuser.so
7fce0b446000-7fce0b447000 r-xp 00000000 103:02 13923715                  /usr/lib/libddsserialization.so
7fce0b447000-7fce0b646000 ---p 00001000 103:02 13923715                  /usr/lib/libddsserialization.soAbgebrochen

```

dirk-thomas, 2016-07-13:
I just tried to build the current state again and it still fails for me to link `ros2_simple_logger_test` as well as `ros2_simple_logger_echo` with the following error message:

```
libros2_simple_logger_lib.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_introspection_cpp::get_message_type_support_handle_introspection<ros2_simple_logger::msg::LoggingMessage_<std::allocator<void> > >()'
collect2: error: ld returned 1 exit status
```

firesurfer, 2016-07-19:
Sry for my late answer. I couldn't find the time over the weekend to deal with this issue. 

Would you mind to take a look at the `CMakeLists.txt`. I tried to build the `ros2_simple_logger::msg::LoggingMessage`  message in the same package as the code that uses this message. I took the necessary cmake commands from the `system_tests` repository. (https://github.com/ros2/system_tests/blob/master/test_rclcpp/CMakeLists.txt) The problem is that I only want to build my code against the default middleware and not against all available middlewares, whereas the system_tests are build for all available middlewares. I'm not sure why this code builds fine on my system but not in yours. 

Perhaps you could provide a simple example on how to build against the default middleware and building messages in the same package as the code that uses them.

firesurfer, 2016-07-19:
I think I was finally able to fix this issue my-self. At least my programs end without any error messages now. The problem was that I was linking multiple middlewares to one binary. Now I'm only linking against the default middleware.

As said before: Perhaps a simple example on how a CMakeLists.txt should look for building your own package with ros2 and ament would be very useful.

dirk-thomas, 2016-07-19:
Can this ticket be closed then?

The current examples are also being used to actually test of the functionality works across all vendors. Therefore they are not as simple as they could be. Please feel free to open a ticket on that repo to request some examples which are closer to what users should write for a single rmw impl.

closed 2016-07-19 by firesurfer