ros2/rclcpp #243: Parameter Events should be node specific

rohbotics, 2016-07-27:
Right now parameter events are published on a global topic `/parameter_events`.

This causes problems when there are multiple nodes listening to `/parameter_events`.
Either the message needs to be changed to contain the node name where the parameter is hosted,
or `/parameter_events` needs to become node local.

I think the local `~/parameter_events` is the cleaner way, as that way callbacks aren't triggered for all nodes every-time there is an update on one node.

wjwwood, 2016-07-27:
I think the purpose of the `/parameter_events` topic is so that a single node can efficiently monitor all changes. It shouldn't be subscribed to by every node that has parameters. However, I do agree that the parameter events should contain the node to which the parameter belongs and possibly also the node that requested the change originally. The former might be addressed if the canonical parameter addresses become `node + parameter namespace + parameter name` as has been discussed before in our meetings.

As for reacting to a parameter changing in a node, i.e. implementing `on_change('param name', function_to_call)`, I think that should be handled in the set parameter code, not by listening to the parameter events.

rohbotics, 2016-07-27:
I think that view of parameter events makes sense. For example a gui showing all the parameters shouldn't have to subscribe to individual node topics.

So for dynamic parameter changes, parameter_service should provide a hook for updating the parameters, as it receives the service call in the first place.
Should the hook take the from of `on_change('param_name', callback)` or `on_param_change(callback)`?

wjwwood, 2016-07-27:
That's correct. In fact further than just having a hook for notifying the node that it has been changed, the hook would allow the node to reject the change if it desired.

I suppose you could have either:

- `node.on_param_change(callback)`
- `node.on_param_change('param_foo', callback)`
- `node.on_param_change('gains/*', callback)`

I think we didn't implement this yet because it is so tied to how we end up formulating parameter names (currently just a variable style name with no namespaces).

rohbotics, 2016-07-28:
@wjwwood I made basic PR for `node.on_param_change(callback)`

sloretz, 2018-01-05:
What is the plan for this ticket? It appears the topic `parameter_events` is currently in the namespace given to the constructor of `Node`, not the global namespace.

https://github.com/ros2/rclcpp/blob/6129a12df5ec21b7ccbf6879036281b1374db0fb/rclcpp/src/rclcpp/node_interfaces/node_parameters.cpp#L41-L46

mikaelarguedas, 2018-01-07:
This will be addressed during the [parameter refactor](https://github.com/ros2/ros2/issues/432).
https://github.com/ros2/rcl_interfaces/pull/27 aims to add the missing information to the paramer event messages so that the full node name is provided.

wjwwood, 2020-03-19:
This is now the case: https://github.com/ros2/rcl_interfaces/blob/93cedce2dacd25fa4f2969d022ccbe3f2903e3fe/rcl_interfaces/msg/ParameterEvent.msg#L7-L8

closed 2020-03-19 by wjwwood