ros2/rclcpp #247: Questions to the rclcpp architecture

firesurfer, 2016-08-22:
I read a lot of code of the rcl and rclcpp the last days, because I'm writing a C# wrapper for ros2 at the moment. I came up with two architectural questions:

1. Why does the rclcpp has an extra intra and interprocess publish and subscribe mechanism. Shouldn't this case be coverd by the DDS implementation ?

2. Why is there an `any_executable` class which can contain a node, subscription and so on? Wouldn't it be easier or convenient to have a executable base class that provides an execute method and which is inherited by publisher, node and so on ?

Thank you in advanced for explaining these design decisions.

dirk-thomas, 2016-08-23:
1. Some DDS implementations provide a shared memory transport but that still requires the data to be serialized into a shared memory segment. On the other hand the intra-process mechanism implemented in `rclcpp` allows for zero copy transport (passing a reference) without the need to serialize the data (similar to ROS 1 nodelets). So it will perform "better". This is specifically useful when running several nodes in a single process which exchange message with high frequency and/or large size.

2. The `any_executable` class uses *aggregation* instead of *inheritance*. It is for sure a subjective choice but imo inheritance is often misused for cases like this. When using inheritance you can only have one class hierarchy and the specific types (like node, subscription) need to be aware of their base class. With aggregation on the other hand these types don't no anything about how they are being used (e.g. in `any_executable`) and it is easy to write multiple different way to aggregate them.

firesurfer, 2016-08-23:
Thank you for the explanations.

closed 2016-08-23 by firesurfer