ros2/rclcpp #248: Timer unreliable on ARM

gerkey, 2016-08-23:
When testing on an rpi3, I found that the `dynamic_bridge` wasn't working. I've tracked it down to the wall timer that's supposed to make the bridge poll the ROS2 graph not firing. I then tried the `timer_test` in `test_rclcpp` and found that it fails intermittently. I'm tempted to blame it on inadequate CPU cycles, but that's not very satisfying.

gerkey, 2016-08-24:
It looks like it can happen that `wait_for_work()` doesn't return even when enough time has elapsed such that a timer definitely should have been ready. It also looks like the timers can work for a while (e.g., a couple of iterations), then stop working altogether for the duration of the process. That's the behavior that I've seen at the application level with `dynamic_bridge`, too.

I guess that whatever is supposed to make the timer wake up `wait_for_work()` isn't working properly in this case. Anybody know where to look for the root cause? FYI, I'm using opensplice.

wjwwood, 2016-08-24:
It could be that our code that calculates when timers are ready is broken, or it could be a race condition between the code that determines how long to wait and asking the DDS implementation to wait (with a timeout), or it could be the DDS implementation from waking up. I can give a few pointers:

This code determines how long to ask the rmw wait (and therefore the DDS wait) to wait for other events, like subscriptions to be ready:

https://github.com/ros2/rcl/blob/e6e7713/rcl/src/rcl/wait.c#L523-L553

This code determines, after waking up from wait, if any timers are ready to be executed:

https://github.com/ros2/rcl/blob/e6e7713/rcl/src/rcl/wait.c#L584-L595

This code does the actual waiting:

https://github.com/ros2/rcl/blob/e6e7713/rcl/src/rcl/wait.c#L555-L562

I find it useful to put print statements throughout that part of the code see if DDS is waking up or not when you expect your timers to be firing but they are not. Also, it's interesting to see if this problem occurs when you also have subscription data coming in periodically.

gerkey, 2016-08-24:
Thanks, @wjwwood. Looks like this line:

https://github.com/ros2/rcl/blob/e6e7713/rcl/src/rcl/wait.c#L538

is sometimes giving back a bogus answer. Digging further.

wjwwood, 2016-08-24:
This line could be a problem since it uses int64 max:

https://github.com/ros2/rcl/blob/e6e7713/rcl/src/rcl/wait.c#L537

Not sure how it might go wrong, but if the arm is not 64-bit or if it emulates 64-bit in a weird way it might be a problem.

rohbotics, 2016-08-24:
@gerkey Turtlebot blue has a ROS2/Xenial arm64 schroot if you want to test on 64 bit.

gerkey, 2016-08-24:
I think that this line:

https://github.com/ros2/rcl/blob/master/rcl/src/rcl/timer.c#L145-L146

can experience rollover of the uint64 value used to represent the current time in nanoseconds. At first I thought that time values were totally bogus, but we're using `clock_gettime()` to get a monotonic clock, which has an arbitrary offset. If the rollover occurs between timer calls, then the calculation of the next time experiences overflow and you get the wrong answer.

At least that's what I think is happening. I'll confirm tomorrow.

gerkey, 2016-08-24:
Btw, this is a 32-bit ARM board, but I don't expect that to be the problem. Surely it can do correct 64-bit integer math?

If my hypothesis about rollover is right, then it could be that we've always had this bug but that there's something about the rpi3 that makes the monotonic clock values more likely to start near the max value and thereby expose the problem.

wjwwood, 2016-08-24:
So you think that the `last_call_time` or `now` is overflowing? I can understand an issue if `last_call_time` is large and `now` rolls over, then you'd have to wait until `now` passes `last_call_time` again before the condition would be true. Is that what you mean?

> Btw, this is a 32-bit ARM board, but I don't expect that to be the problem. Surely it can do correct 64-bit integer math?

Yeah, I didn't expect it to be an issue.

Can you imagine a test that could expose this flaw in our logic? That might be easier than relying on getting a monotonic time that triggers it.

gerkey, 2016-08-24:
I'm pretty sure that I saw `now` rollover. If that happens, then `last_call_time + period` is a very large number, while `now` is small, and we'll decide that the time until the next call is very long. On the other hand, I also saw a situation in which `now` was very large and `last_call_time` was very small, and I can't explain that.

I'll try tomorrow to reproduce it.

As for testing, I guess that we first need to implement something like a `rollover_proof_uint64_ns_time_plus()` helper function that can be tested, and then we start using that.

dirk-thomas, 2016-08-24:
If I didn't check the wrong thing the function `rcl_timer_get_time_until_next_call` seems to work just fine in the case of a steady time overflow: fede5e1c9fcb5f52751140d9f9d81449e7752f5f

But I only ran the test on my machine. Maybe the result is different on the ARM board?

gerkey, 2016-08-24:
Thanks for the test! It passes for me. But I did initially get one warning and one error during compilation:
~~~
/home/pi/ros2_ws/src/ros2/rcl/rcl/src/rcl/timer.c: In function ‘rcl_timer_exchange_callback’:
/home/pi/ros2_ws/src/ros2/rcl/rcl/src/rcl/timer.c:201:10: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   return (rcl_timer_callback_t)rcl_atomic_exchange_uintptr_t(
          ^
/home/pi/ros2_ws/src/ros2/rcl/rcl/src/rcl/timer.c: In function ‘_rcl_timer_set_last_call_time’:
/home/pi/ros2_ws/src/ros2/rcl/rcl/src/rcl/timer.c:244:31: error: incompatible types when assigning to type ‘atomic_uint_least64_t’ from type ‘rcl_time_point_value_t’
   timer->impl->last_call_time = value;
                               ^
~~~

I fixed the error with this change:
~~~
diff --git a/rcl/src/rcl/timer.c b/rcl/src/rcl/timer.c
index 1d9893a..b22d1e3 100644
--- a/rcl/src/rcl/timer.c
+++ b/rcl/src/rcl/timer.c
@@ -241,7 +241,7 @@ _rcl_timer_set_last_call_time(rcl_timer_t * timer, rcl_time_point_value_t value)
 {
   RCL_CHECK_ARGUMENT_FOR_NULL(timer, RCL_RET_INVALID_ARGUMENT);
   RCL_CHECK_FOR_NULL_WITH_MSG(timer->impl, "timer is invalid", return RCL_RET_TIMER_INVALID);
-  timer->impl->last_call_time = value;
+  rcl_atomic_store(&timer->impl->last_call_time, value);
   return RCL_RET_OK;
 }
~~~

Are these issues perhaps symptomatic of the underlying problem?

Also, I'll dig into the test, to be sure that it's testing the case that I think I'm seeing.

wjwwood, 2016-08-24:
The build error means you probably have a different implementation of the C11 atomics than what comes with our systems. On ours, the uint64 atomic is just an int, but it seems to be something else, perhaps a structure. The latter is allowed by the C11 atomics interface. That's why they have macros for assignment and initialization and stuff. You patch is more correct, it just happened to work on our machines are @dirk-thomas wrote it because of how our atomics are implemented. Which version of gcc are you using?

gerkey, 2016-08-24:
The problem (at least the biggest problem) is in the underlying time functions of `rcl`: ros2/rcl#75.

gerkey, 2016-08-24:
Btw, @dirk-thomas, pushed a compile fix into your branch to add the other overflow test: cc0256c280ba4e9ded39c091d508900d8e5da75b. Will you make a PR or should I?

closed 2016-08-24 by gerkey

dirk-thomas, 2016-08-24:
I created ros2/rcl#76 but I am not convinced it should be merged.

gerkey, 2016-08-25:
I'm very happy to report that with ros2/rcl#75 in place, `dynamic_bridge` is now working on the rpi3.