ros2/rclcpp #263: [connext] take request failed for server of service

dhood, 2016-10-28:
Client-server examples for Connext seem to be functioning fine but also outputting the following error:

```
$ add_two_ints_server__rmw_connext_cpp 
RTI Data Distribution Service EVAL License issued to SE dhood@osrfoundation.org For non-production use only.
Expires on 02-Nov-2016 See www.rti.com for more information.
Incoming request
a: 2 b: 3
[rclcpp::error] take request failed for server of service 'add_two_ints': error not set, at /home/dhood/ros2_connext168/src/ros2/rcl/rcl/src/rcl/service.c:172
```

http://ci.ros2.org/view/nightly/job/nightly_linux_release/242/testReport/(root)/test_client_scope_cpp__rmw_connext_cpp_Release/test_client_scope_cpp/

dhood, 2016-11-10:
I don't think ros2/rcl#88 is enough to fix this error message (the error is still output on the CI from that PR: http://ci.ros2.org/job/ci_linux/1908/testReport/(root)/test_add_two_ints_server_add_two_ints_client__rmw_connext_cpp_/test_executable/)

We would also have to change the behaviour on `RCL_RET_SERVICE_TAKE_FAILED` in rclcpp: https://github.com/ros2/rclcpp/blob/f28bb1107881a983efef78d5022a257e3b09eda4/rclcpp/src/rclcpp/executor.cpp#L299

What was the original motivation for the `RCL_RET_SERVICE_TAKE_FAILED` return code? If it's not appropriate to raise an error, what is appropriate?

dirk-thomas, 2016-11-10:
Sorry for closing this too fast. #271 should address the remaining issue.

mikaelarguedas, 2016-11-10:
My understanding was that if you receive OK you process the message while if you receive TAKE_FAILED the call didnt cause an error but there is no message to process and relay to the higher layer of the stack.
Processing the error message the same way as it is don for [rcl_take](https://github.com/ros2/rclcpp/blob/f28bb1107881a983efef78d5022a257e3b09eda4/rclcpp/src/rclcpp/executor.cpp#L248) should allow a more consistent behavior

dirk-thomas, 2016-11-10:
The new build doesn't have the error message anymore: http://ci.ros2.org/job/ci_linux/1910/testReport/(root)/test_add_two_ints_server_add_two_ints_client__rmw_connext_cpp_/test_executable/

closed 2016-11-10 by dirk-thomas