ros2/rclcpp #264: has_invalid_weak_nodes can only ever be false

gbiggs, 2016-10-31:
In allocator_memory_strategy.hpp in the function `collect_entities`, the variable `has_invalid_weak_nodes` looks to me like it can only ever be `false`.

https://github.com/ros2/rclcpp/blob/7494350ad24201bd0ef6f75b94760830948274cb/rclcpp/include/rclcpp/strategies/allocator_memory_strategy.hpp#L143

Is the correct behaviour for line 147 to set the value to `true`?

wjwwood, 2016-11-01:
I think this line is wrong:

https://github.com/ros2/rclcpp/blob/7494350ad24201bd0ef6f75b94760830948274cb/rclcpp/include/rclcpp/strategies/allocator_memory_strategy.hpp#L147

It should be `= true` here.

wjwwood, 2016-11-01:
Think you could open a pr that fixes that? I think it means that essentially no invalid nodes are ever removed from this list:

https://github.com/ros2/rclcpp/blob/f28bb1107881a983efef78d5022a257e3b09eda4/rclcpp/src/rclcpp/executor.cpp#L331-L346

PR #266 (2016-11-17): Fix has_invalid_weak_nodes
Fixes #264/

My copy of the master for ROS 2 doesn't currently compile so I worked against Alpha8. The changes apply cleanly to master, however.

The need to call `rclcpp::init()` in the tests so that nodes can be created is a bit ugly. I'd welcome suggestions on how to improve that. (Making `Node` an interface so that mocks can be easily created would be a good approach, but also a significant change to the code base; I'm happy to do that work if there is interest.)
commit aa2d0a39 (2016-11-17): Fix has_invalid_weak_nodes (#266)

* Fix #264

* Corrected test comment
commit c30a020b (2016-11-18): Fix has_invalid_weak_nodes (#266)

* Fix #264

* Corrected test comment
closed 2016-11-17 by dhood