ros2/rclcpp #267: create_service not workin in some cases

firesurfer, 2016-11-03:
I pulled yesterday the lasted code from all master branches. Since then I got the problem that in one case the create_service call from the node class isn't working properly. I got the following code in the constructor of a class:

```c++
 using namespace std::placeholders;

    mac_service = _parentNode->create_service<some_service_msg>("CubePort"+std::to_string(getId())+"_srv", std::bind(&CubePort::HandleMacServiceRequest,this,_1,_2,_3));
    
//This is added for debugging
    std::map<std::string,std::string> res = _parentNode->get_topic_names_and_types();
    for(auto it= res.begin(); it != res.end(); it++) {
        std::cout<< it->first << " : " << it->second << std::endl;
       
    }
```
I can't access the created service and the map given by `get_topic_names_and_types` doesn't list the necessary publisher and subscriptions needed for the service. I checked by modifing the service example that this list does also contain publisher and subscription created for services.
Please also note that other services in the same application are created and listed correctly. 

Currently I'm stuck with debugging this problem because when I tried recreating such a scenario I couldn't reproduce this error. Could you point out some cases where the creation of a service could fail without any error message?

firesurfer, 2016-11-07:
I got it working. Apparently this behavior occurs if there are two services with the same name but different types.

firesurfer, 2016-12-01:
Perhaps it would be great to implement a check in the rclcpp or the rcl if a service of the same name but different type has already been added.

wjwwood, 2017-01-28:
That would definitely be an issue. We've still not decided whether or not services need to be unique (i.e. one service server per name, regardless of type).

closed 2023-07-17 by firesurfer