ros2/rclcpp #273: no std::bind with custom arguments for callbacks

Karsten1987, 2016-11-16:
I want to pass custom arguments to my callback function and use std::bind for it. However, I believe that function_traits are not allowing passing a functor with custom arguments.

This little demo shows the block which does not compile. 
https://github.com/ros2/examples/blob/callback_error/rclcpp_examples/src/services/add_two_ints_server.cpp#L58-L68

Karsten1987, 2016-11-16:
@esteve can you provide some help here? I guess that https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/function_traits.hpp doesn't allow additional arguments?

esteve, 2016-11-17:
@Karsten1987 sure, I just submitted #278 which should work with your example, it's just a quick hack. There's one caveat, though, you can't wrap the `std::bind` in a `std::function` (https://github.com/ros2/examples/blob/callback_error/rclcpp_examples/src/services/add_two_ints_server.cpp#L61), you have to pass the `std::bind` to `create_service` as is, like this:

```c++
auto fcn2 = std::bind(handle_add_two_ints2, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3, i);
node->create_service<example_interfaces::srv::AddTwoInts>("add_two_ints", fcn2);
```

I haven't tested the pull request heavily, so I can't guarantee that it will work with all cases, but it should work with yours :-)

esteve, 2016-11-18:
@Karsten1987 I think this would work, without changing anything in `rclcpp`:

```cpp
std::function<void(const std::shared_ptr<rmw_request_id_t>,
  const std::shared_ptr<example_interfaces::srv::AddTwoInts::Request>,
  std::shared_ptr<example_interfaces::srv::AddTwoInts::Response>)> fcn2;
fcn2 = std::bind(handle_add_two_ints2, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3, i);
node->create_service<example_interfaces::srv::AddTwoInts>("add_two_ints", fcn2);
```

note that I removed the fourth argument in the `std::function` definition. `std::bind` can be wrapped in a `std::function` even though their parameters don't match, as long as when you call it the call signature is compatible. When the wrapped `std::bind` gets called, it'll be compatible since it only has three placeholders, and `i` is already known at the time the `std::bind` was constructed.

Karsten1987, 2016-11-18:
@esteve thanks a bunch for iterating on this. Your solution(s) definitely do the trick.

The fact that you can't easily store them in a `std::function` may become a bit inconvenient at one point; In case you want to store functors in a callback group. Despite the fact that you can't use `auto` that easily.  Is there anyway around this caveat?

In any case, that works just fine and we can simply document this behavior in order to avoid any confusion.

esteve, 2016-11-30:
@Karsten1987 you can store `std::bind` in a `std::function` without #278, as long as:

- the `std::bind` uses at most 3 placeholder arguments
- the placeholder arguments appear sequentially. That is, `std::bind(func, std::placeholder::_2, std::placeholder::_1)` won't work

so given that, this should be fine with the current codebase without any changes to `function_traits.hpp`:

```cpp
int i = 42;
std::function<void(const std::shared_ptr<rmw_request_id_t>,
  const std::shared_ptr<example_interfaces::srv::AddTwoInts::Request>,
  std::shared_ptr<example_interfaces::srv::AddTwoInts::Response>)> fcn2 = std::bind(
    handle_add_two_ints2, std::placeholders::_1, std::placeholders::_2,
    std::placeholders::_3, i);
node->create_service<example_interfaces::srv::AddTwoInts>("add_two_ints", fcn2);
```

codebot, 2016-12-13:
In interest of getting Beta 1 out the door today, this is deferred to Beta 2.

Karsten1987, 2017-08-10:
@esteve thanks a bunch for the work you've put into it.
As we currently don't have a real show-stopper with it, i'll go ahead and close it. We can open or refer to it when we actually start dealing with MSVC2017.
Cheers!

closed 2017-08-10 by Karsten1987