ros2/rclcpp #281: Subscription callback should support const reference by default

geoffviola, 2016-11-29:
I noticed that the any_subscription_callback supports shared_ptr and unique_ptr, but not const&: https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/any_subscription_callback.hpp. It seems simpler to have const& on the subscription callback. The user almost always uses the data in the callback scope. Generally, he will not change the owner of the message data to avoid copying. The rclpcc example should use it: https://github.com/ros2/examples/blob/master/rclcpp_examples/src/topics/listener.cpp.

wjwwood, 2017-01-28:
Yeah, I think it makes sense to support const reference, but never got around to doing. That would be a good enhancement. As for whether or not to use it by default, I think that might also be the right thing to do, but we were trying to stay as close as possible to ROS 1's simple examples (not a reason to not change, but just a historical reason why it is that way right now):

http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29#roscpp_tutorials.2BAC8-Tutorials.2BAC8-WritingPublisherSubscriber.CA-b00794bcdcacff4e742e844261078a50000d0a69_34

henningkayser, 2020-06-12:
@wjwwood any updates on this issue?  It seems like it would be really easy to add const-ref callbacks and I think it's a very reasonable default. With ROS1 const-ref callbacks were also available and we use them a lot in MoveIt.

wjwwood, 2020-06-13:
Nope, no one has had time. It would be a good enhancement though. We need to audit the allowed signatures as well. We did this at some point for publish and deprecated several different styles and added some others.

wjwwood, 2021-07-16:
Closed by #1557

closed 2021-07-16 by wjwwood