ros2/rclcpp #282: Warning against defining user-defined literals in the global namespace

Sarcasm, 2016-11-29:
Hello,

I was just giving a small glimpse at the code, and I noticed that some user-defined literals are not part of the rclcpp namespace.

- [rclcpp/include/rclcpp/rclcpp.hpp#L31-L34](https://github.com/ros2/rclcpp/blob/a987f8d01515562f5727158eb25a957150db4e45/rclcpp/include/rclcpp/rclcpp.hpp#L31-L34)

I think it is dangerous, and as a user of the library, I would prefer that the library does not pollute the global namespace.

For example, imagine another library decide to do the same, but using the "Possible implementation" definition of the cppereference.com documentation for `operator ""s`?

- http://en.cppreference.com/w/cpp/chrono/operator%22%22s#Possible_implementation

It means we can end up in the following situation:

**rclcpp.hpp**

```cpp
...
inline const std::chrono::seconds operator"" _s(unsigned long long s)
{
  return std::chrono::seconds(s);
}
...
```

**foo.hpp**
```cpp
...
constexpr std::chrono::seconds operator ""_s(unsigned long long s)
{
    return std::chrono::seconds(s);
}
...
```

**my_node.cpp**
```cpp
#include <foo.hpp>
#include <rclcpp/rclcpp.hpp>

int main()
{
    std::cout << (30_s).count() << " seconds\n";
    return 0;
}
```

The code does not compile, live demonstration available here:

- http://coliru.stacked-crooked.com/a/aabea17bb76b91fd

In the standard library the chrono literals are in the namespace `std::chrono_literals`.
By symmetry I guess the rclcpp ones could be defined in `rclcpp_literals`, but I guess being under the rclcpp namespace makes even more sense, e.g: `rclcpp::literals`, only one namespace (and sub-namespaces) for the library.

wjwwood, 2016-11-29:
Yeah, this was sortof a temporary solution because C++14 was widely available at the time and the chrono literals were only added in C++14. We should consider removing this in favor of the literals provided by C++14 (http://en.cppreference.com/w/cpp/chrono/operator%22%22s). Now that C++14 is more widely available, we'll likely make it a requirement, unless we run into something that blocks us from doing so.

I didn't put them in a namespace because I didn't want to clutter the minimal user examples, but you're right that if we keep them, we should do something like put them in a namespace and make the user do this:

```c++
#include <rclcpp/rclcpp.hpp>

using namespace rclcpp::literals;

int main(...) {
  auto timer = node->create_timer(10_s, [](){});
}
```

Sarcasm, 2016-11-30:
Ok, thank you.

Let me know if a pull request interest you, that would be a good excuse for me to take a look at ros2.

wjwwood, 2016-11-30:
We're always happy to see a pull request. If you're interested in making one, I'd suggest putting the literals in the `rclcpp::literals` namespace, perhaps in their own `rclcpp/literals.hpp` file which would be included in `rclcpp/rclcpp.hpp`, and then either do `using namespace rclcpp::literals` in `rclcpp/rclcpp.hpp` to preserve existing behavior or go and update all of the examples and demos in various repositories that would be affected by this change. If you just move them into a namespace and then do the `using namespace` to preserve the existing behavior, then we can do this in two steps, one that only touches this repository and then a second which touches all of the code out there using the `rclcpp::literals` implicitly.

wjwwood, 2016-11-30:
This would also be a good opportunity to consider using the `constexpr` like the "possible implementation" example on the cppreference website.

dirk-thomas, 2016-11-30:
If the C++14 `chrono_literals` work on all current platforms (Ubuntu Xenial, Mac OS, Windows using VS 2015 Update 3) it would be good to use them.

wjwwood, 2016-11-30:
That's fine too, but we have to make an explicit decision to require that version of C++, which means making an announcement, updating the wiki pages, and updating the `-std` statements in all packages. That's probably not something an external contributor could do on their own, but @Sarcasm is welcome to open prs to help get the ball moving on that too, as an alternative to what I described above.

Sarcasm, 2016-11-30:
I will take a look at the first proposition for now.

Regarding the switch to C++14, a quick look seems to indicate that it should be possible for the 3 platforms you listed.
Apparently the chrono literals are available since GCC 5, Clang 3.4 and MSVC 14

- http://en.cppreference.com/w/cpp/compiler_support

I have to get more familiar with the CMakeLists.txt of the ros2 packages but I can already imagine a few patches to propose which could ease making the switch without breaking the current state of things.

wjwwood, 2016-12-17:
@Sarcasm, since we decided (and executed) to make the switch to c++14, I'm going to touch each of your pull requests to use the `std::chrono_literals` throughout and remove the literals defined in `rclcpp`.

wjwwood, 2016-12-17:
Thanks for contribution @Sarcasm, sorry for the delay while we sorted out the c++14 stuff.

closed 2016-12-17 by wjwwood

Sarcasm, 2016-12-17:
No problem I understand the decision was an important one, I'm glad you made the switch.
Thank you for the taking the time to look at the PRs.