ros2/rclcpp #295: Parameter Service not started by default in Node

rohbotics, 2016-12-22:
I can't find when we stopped doing this, (or if we ever did), but the parameter service should be started along with the Node. 

There might be some circular references, as ParameterService stores a shared_ptr of Node, but it should work with weak_ptr.

Some discussion here:
https://discourse.ros.org/t/composition-and-parameters-best-practice-suggestions/1001

ahcorde, 2016-12-29:
Adding this line ` auto parameter_service = std::make_shared<rclcpp::parameter_service::ParameterService>(node);` behind the instance of your node should be enough.

```
  auto node = rclcpp::Node::make_shared("listener");
  auto parameter_service = std::make_shared<rclcpp::parameter_service::ParameterService>(node);
```

**NOTE:**
```
  // TODO(esteve): Make the parameter service automatically start with the node.
```
See this comments in the code: [1](https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/parameters/set_and_get_parameters.cpp#L26) or [2](https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/parameters/parameter_events.cpp#L46)

wjwwood, 2017-01-03:
@ahcorde is right, I remember this now. We didn't want to turn it on by default until there was an option to turn it off as well.

clalancette, 2018-10-26:
We actually did this earlier this year for the bouncy release, so I'm going to close this out.

closed 2018-10-26 by clalancette