ros2/rclcpp #305: Open Client API for inheritance

AravindaDP, 2017-02-10:
Hi
Like in https://github.com/ros2/rclcpp/pull/258 where Node/Publisher API is opened for inheritance shouldn't we open the client API also for inheritance?

Probably the reason behind Node/Publisher API being allowed to inherit may not apply Client API. Anyway the reason I do this suggestion is for the mocking/unit testing perspective.

For example. Now I could pass in a publisher into my node class and mock publish() calls of the said publisher in the unit tests of my node class. (To verify whether my node class correctly called the corresponding methods in the publisher interface). So if we had the async_send_request call
https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/client.hpp#L239
in the Client API also marked as virtual we could extend the same capability for the client objects as well.

I should admit that my experience with C++ and the ros2 core is limited. So what I propose above may not be the correct way to achieve the said requirement. Or there may be strong reasons against that warrant such a change.

If the proposed change is acceptable way to do this I would be more than happy to try to implement it and open a pull request.

Please let me know if any additional information is required.

wjwwood, 2017-02-27:
@AravindaDP Sorry for the delay in responding. Yes! that is the reason we decided to break up the Node class into pure virtual components, so that we could do mocking more easily and so that we could build different node classes with different API's exposed.

If you'd like to try and do a similar thing to Service servers/clients and/or publishers and subscribers then we'd be happy to try and integrate it. I'd recommend, however, that you try and express your idea for how the classes will be laid out before trying to make a final pull request though, that way we can give you feedback on the proposed design. If you think it's not that much work you can open the pr before getting feedback, but know that we might ask for changes before merging.

Thanks!

wjwwood, 2017-02-27:
Also, if you'd like to just propose one or more methods be made `virtual`, you can do that too. We haven't really reviewed that part of the API very closely yet, so it's quite possible it totally makes sense and we just overlooked it the first time through.

Karsten1987, 2018-02-26:
@AravindaDP I am closing this ticket due to inactivity.
Thanks for bringing the topic forward and please feel free to re-open the ticket in case there's any update.

Karsten1987, 2018-02-26:
I am re-opening this ticket for visibility but add it to the untargeted milestone. @AravindaDP, this means we won't spend time on it in the near future, but again please feel free to bring forward any concern you might face with this issue.