ros2/rclcpp #308: SyncParametersClient prevents future publishers and services from being created

wjwwood, 2017-03-01:
I've got a really strange situation where anything (in this case publishers and service servers) created after using the `SyncParametersClient` does not show up on the network. The workflow is as follows:

- create node
- create parameter service
- create publisher on `taco`
- create `SyncParametersClient` and use it
- create publisher on `map`

The observed behavior is that the parameter service's topics and `taco` shows up in the `rostopic_list`'s output, but `map` does not. If I comment out the creation of and usage of the `SyncParametersClient`, then map shows up.

I'm going to try and make a test to expose this issue.

wjwwood, 2017-03-01:
To clarify, the publishers and service servers can be created successfully (no exception and objects are valid afterwards) and you can call `.publish()` on the publishers, but they will never show up in `rostopic_list` (which uses `get_topic_names_and_types`) and a subscriber of those topics (C++ or Python) will not receive any data (even if both are using default QoS settings). The topics do show up in `rostopic_list` once you create the subscription.

My guess is that it puts the node in some funky state because of how the sync client spins to get the service client response, but honestly I find it completely bizarre.

Karsten1987, 2018-02-26:
@wjwwood Do you think this is still an open issue?
I'll assign you to this ticket and add it to the bouncy milestone. I guess, it'd be great if we could verify that this behavior is still the same with the current ros2 master. If so, we can tackle it for the next release.
Feel free to re-assign the ticket.

wjwwood, 2018-03-20:
I was unable to reproduce this with this code using master and fastrtps:

```c++
#include <memory>

#include "rclcpp/rclcpp.hpp"

#include "std_msgs/msg/string.hpp"

using namespace std::chrono_literals;

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);

  auto node = std::make_shared<rclcpp::Node>("issue_308");

  auto param_server = std::make_shared<rclcpp::ParameterService>(node);

  auto taco_pub = node->create_publisher<std_msgs::msg::String>("taco");

  auto param_client = std::make_shared<rclcpp::SyncParametersClient>(node);

  auto map_pub = node->create_publisher<std_msgs::msg::String>("map");

  rclcpp::spin(node);

  return 0;
}
```

closed 2018-03-20 by wjwwood