ros2/rclcpp #312: Calling a Service from an other Service leading to a std::runtime_error

PhilKW, 2017-03-21:
I need to call a service from a service call, which is embedded in a Node class. 
To make the problem clear I adapted the Add two Ints example:
Node 1 provides the standard Add two Ints service.
```
#include <inttypes.h>
#include <memory>
#include "example_interfaces/srv/add_two_ints.hpp"
#include "rclcpp/rclcpp.hpp"

using AddTwoInts = example_interfaces::srv::AddTwoInts;

void handle_service(
  const std::shared_ptr<rmw_request_id_t> request_header,
  const std::shared_ptr<AddTwoInts::Request> request,
  const std::shared_ptr<AddTwoInts::Response> response)
{
  (void)request_header;
  printf("request: %" PRId64 " + %" PRId64 "\n", request->a, request->b);
  response->sum = request->a + request->b;
}

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);
  auto node = rclcpp::Node::make_shared("minimal_service");
  auto server = node->create_service<AddTwoInts>("add", handle_service);
  rclcpp::spin(node);
  return 0;
}
```

Node 2 provides a second service, which calls the Add two Ints service from Node 1 when called.
```
#include <chrono>
#include <cinttypes>
#include <memory>
#include "example_interfaces/srv/add_two_ints.hpp"
#include "rclcpp/rclcpp.hpp"

using AddTwoInts = example_interfaces::srv::AddTwoInts;

using std::placeholders::_1;
using std::placeholders::_2;
using std::placeholders::_3;

using namespace std::chrono_literals;

class MinimalClient : public rclcpp::Node
{
public:
    MinimalClient(): Node("minimal_client")
    {
        
    }
    
    void init()
    {
        client = this->create_client<AddTwoInts>("add");
        
        rclcpp::service::Service<AddTwoInts>::SharedPtr server = 
                this->create_service<AddTwoInts>("add2", 
                std::bind(&MinimalClient::handle, this, _1, _2, _3));
        
        rclcpp::spin(shared_from_this());
    }
        
    void handle(const std::shared_ptr<rmw_request_id_t> /*request_header*/,
        const std::shared_ptr<AddTwoInts::Request> request_2,
        std::shared_ptr<AddTwoInts::Response> response_2)
        {        
        while (!client->wait_for_service(std::chrono::seconds(1))) {
            if (!rclcpp::ok()) {
            printf("client interrupted while waiting for service to appear.\n");
            //return 1;
            }
            printf("waiting for service to appear...\n");
        }
        auto request = std::make_shared<AddTwoInts::Request>();
        request->a = request_2->a;
        request->b = request_2->b;
        auto result_future = client->async_send_request(request);
        std::cout<<"1"<<std::endl;
        std::shared_ptr<rclcpp::Node> parent = shared_from_this();
        std::cout<<"2"<<std::endl;
        if (rclcpp::spin_until_future_complete(parent, result_future) !=    rclcpp::executor::FutureReturnCode::SUCCESS)
        //if (result_future.wait_for(5s)!= std::future_status::ready)
        {
            printf("service call failed :(\n");
            return;
            //return 1;
        }
        auto result = result_future.get();
        printf("result of %" PRId64 " + %" PRId64 " = %" PRId64 "\n",
            request->a, request->b, result->sum);
        response_2->sum = result -> sum;
        //return 0;
    }
private:
    rclcpp::client::Client<AddTwoInts>::SharedPtr client;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  auto ptr = std::make_shared<MinimalClient>();
  ptr -> init();
  rclcpp::spin(ptr);
  return 0;
}
```

Node 3 is the standard client from the Add two Ints example calling the service from Node 2.

```
#include <chrono>
#include <cinttypes>
#include <memory>
#include "example_interfaces/srv/add_two_ints.hpp"
#include "rclcpp/rclcpp.hpp"

using AddTwoInts = example_interfaces::srv::AddTwoInts;

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  auto node = rclcpp::node::Node::make_shared("minimal_client");
  auto client = node->create_client<AddTwoInts>("add2");
  while (!client->wait_for_service(std::chrono::seconds(1))) {
    if (!rclcpp::ok()) {
      printf("client interrupted while waiting for service to appear.\n");
      return 1;
    }
    printf("waiting for service to appear...\n");
  }
  auto request = std::make_shared<AddTwoInts::Request>();
  request->a = 41;
  request->b = 1;
  auto result_future = client->async_send_request(request);
  if (rclcpp::spin_until_future_complete(node, result_future) !=
    rclcpp::executor::FutureReturnCode::SUCCESS)
  {
    printf("service call failed :(\n");
    return 1;
  }
  auto result = result_future.get();
  printf("result of %" PRId64 " + %" PRId64 " = %" PRId64 "\n",
    request->a, request->b, result->sum);
  return 0;
}
```

The problem occurs due to the `spin_until_future_complete` function adding the node to the spin, when it is already spinning. (In my specific problem Node 2 also provides several topics so the spin function is needed) The Error Message is:

> terminate called after throwing an instance of 'std::runtime_error'
> what():  Node has already been added to an executor.
> Aborted (core dumped)

I figured that a simple `result_future.wait_for()` should do it but it never gets triggered that the future is ready. Same goes for `result_future.wait()` and `result_future.wait_until()`

Any suggestions how to work around the `spin_until_future_complete`?
This issue could be related to the issue #206 or #209.

dhood, 2018-03-30:
@PhilKW sorry that this ticket was not on our radar, I just stumbled upon it today. You probably have already found the solution to your issue but for posterity:

Instead of using `spin_until_future_complete` in Node 2 (since the node is already being spun on), you can send the async request with a callback that will get triggered when it is completed. [This demo](https://github.com/ros2/demos/blob/40e3732791c8c72215a5eb954e9e25be1a5ebb73/demo_nodes_cpp/src/services/add_two_ints_client_async.cpp#L62..L73) has an example of how to do that:

```
    // We give the async_send_request() method a callback that will get executed once the response
    // is received.
    // This way we can return immediately from this method and allow other work to be done by the
    // executor in `spin` while waiting for the response.
    using ServiceResponseFuture =
      rclcpp::Client<example_interfaces::srv::AddTwoInts>::SharedFuture;
    auto response_received_callback = [this](ServiceResponseFuture future) {
        auto result = future.get();
        RCLCPP_INFO(this->get_logger(), "Result of add_two_ints: %" PRId64, result->sum)
        rclcpp::shutdown();
      };
    auto future_result = client_->async_send_request(request, response_received_callback);
```

I will close this for now under the assumption that you have sorted this out in the meantime, but if you have other questions feel free to comment here and we can re-open this.

closed 2018-03-30 by dhood

jdlangs, 2019-09-23:
@dhood are you aware if anything has changed in the last ~18 months to make this possible? Calling a service from the callback of another service is *very* common in the code bases I've seen/worked with. Having to arrange another layer of callbacks that somehow passes information back to the original service handler (which your async example does not demonstrate) can be quite a dramatic increase in complexity compared to a synchronous version.

jacobperron, 2019-10-03:
@jdlangs I may be mistaken, but I believe this issue is related to a limitation of the executor as described in https://github.com/ros2/rclcpp/issues/773. In order to call a service from within  another services callback in a synchronous manner would require being able to "spin" recursively.