ros2/rclcpp #315: Non periodic timer

ArkturusMengsk, 2017-03-28:
I am in need of a non periodic timer, that I can start and stop multiple times.
The ros2 timer created with `create_wall_timer` inside a node is naturally periodic, which is not a problem, if it would be possible to stop the timer and start it again later.

There is the possibility to `cancel` the timer, but no way to start it all over again.
`cancel` itself calls `rcl_timer_cancel` from rcl. There the documentation says that you can reset the timer with `rcl_timer_reset` which needs only the timer handle. But the handle I get with `TimerBase::get_timer_handle` is const so I can't use this to reset the timer. 

`rclcpp::TimerBase::SharedPtr myTimer;  // As class member`
`myTimer = this->create_wall_timer(...);  // Class derived from node`
`myTimer->cancel();`
`rcl_timer_reset(myTimer->get_timer_handle());  // does not work`

Another solution would be to create a timer object, use it for one single period, halt and delete it and create another timer object the next time I need one. But with the creation of the timer (`create_wall_timer`) it is added to the internal list of timers `node_timers_`. Unfortunately there is no way to delete the timer from the list!

`this->get_node_timers_interface()->..  // Can only add new timer`

So how the timers in ros2 are meant to be used?
Is there a way I am missing?
Do I really have to write my own timer based on std::clock?

wjwwood, 2017-05-09:
High @ArkturusMengsk, sorry it took so long for me to get back to you. You're not crazy, it's intended that both the use cases you described ought to work, either:

- Start a timer, cancel it, start it again later or
- Start a timer, delete it (implies removing it from any lists of timers), and create a new one later

I'll have to have a look at the code to see what we've missed. As you pointed out there's probably an interface in there that requires non-`const` but you only have access to `const`, and likely there is a bug or missing API to fully remove the timers.

Sorry for the inconvenience, I'll try to have a look what API's need to change or be extended to support these two cases. If in the meantime you've worked around this let me know, maybe I can integrate your changes into the mainline as a contribution.

Sorry again for the delay in responding.

wjwwood, 2017-05-09:
For at least the second case (canceling, deleting and replacing with a new timer) it appears to work. I looked into how the timers are stored in node. As I thought (though I didn't suggest it in my previous post because I wasn't sure yet) the `TimerBase` shared pointer returned by `rclcpp::Node::create_wall_timer` is _scoped_, meaning that when you delete the shared pointer returned to you, it is removed from the internal storage in the Node. The timer is actually stored in a callback group, but that's not important because the callback group is owned by the node. You can see here that it is stored as a weak pointer:

asdfasdf

This is the code I used to test this use case and it works as I would expect:

```c++
#include <chrono>
#include <cstdio>
#include <memory>

#include "rclcpp/rclcpp.hpp"

using namespace std::chrono_literals;

class OneOffTimerNode : public rclcpp::Node
{
public:
  OneOffTimerNode() : rclcpp::Node("one_off_timers"), count(0) {
    periodic_timer = this->create_wall_timer(2s, [this]() {
      printf("in periodic_timer callback\n");
      if (this->count++ % 3 == 0) {
        printf("  resetting one off timer\n");
        this->one_off_timer = this->create_wall_timer(1s, [this]() {
          printf("in one_off_timer callback\n");
          this->one_off_timer->cancel();
        });
      } else {
        printf("  not resetting one off timer\n");
      }
    });
  }

  rclcpp::TimerBase::SharedPtr periodic_timer;
  rclcpp::TimerBase::SharedPtr one_off_timer;
  size_t count;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);

  auto node = std::make_shared<OneOffTimerNode>();

  rclcpp::spin(node);

  return 0;
}
```

You can see my callback for the "one off timer" cancels itself so it is not called again. In the future (on `count`'s that are a multiple of three) it is replaced with a new timer.

I'll look into reusing a single timer every once in awhile by resetting it rather than deleting it and creating a new timer.

It may also be the case that we should have a special "one off" timer object that meets this use case specifically rather than having a periodic timer that cancels itself in the callback.

wjwwood, 2017-05-09:
I tested the other use case of reusing a timer over and over again, and indeed there is functionality in `rcl` (the core C api) that was not exposed in our C++ api. I opened this pull request to add a method for resetting the timer in C++: https://github.com/ros2/rclcpp/pull/326

With that I can run this code successfully:

```c++
#include <chrono>
#include <cstdio>
#include <memory>

#include "rclcpp/rclcpp.hpp"

using namespace std::chrono_literals;

class OneOffTimerNode : public rclcpp::Node
{
public:
  OneOffTimerNode() : rclcpp::Node("reuse_timer"), count(0) {
    one_off_timer = this->create_wall_timer(1s, [this]() {
      printf("in one_off_timer callback\n");
      this->one_off_timer->cancel();
    });
    // cancel immediately to prevent it running the first time.
    one_off_timer->cancel();

    periodic_timer = this->create_wall_timer(2s, [this]() {
      printf("in periodic_timer callback\n");
      if (this->count++ % 3 == 0) {
        printf("  resetting one off timer\n");
        this->one_off_timer->reset();
      } else {
        printf("  not resetting one off timer\n");
      }
    });
  }

  rclcpp::TimerBase::SharedPtr periodic_timer;
  rclcpp::TimerBase::SharedPtr one_off_timer;
  size_t count;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);

  auto node = std::make_shared<OneOffTimerNode>();

  rclcpp::spin(node);

  return 0;
}
```

With those two examples and the pull request I linked to I think it should meet your requirements.

Can you see if any of this helps you? Thanks!

wjwwood, 2017-05-09:
Btw, I put those two examples in our demo package too: https://github.com/ros2/demos/pull/127

ArkturusMengsk, 2017-05-09:
Thank you for attending the issue.
I really missed the self removal from all lists and searched for a way to do this manually. Clearly my fault.

The way you fixed the second part with opening the reset method to the C++ world, looks good to me. This is exactly, what I was missing.
I'm not able to test your changes right now, but I will do this tomorrow.

Thank you!

closed 2017-05-10 by wjwwood