ros2/rclcpp #317: No shutdown after SIGINT

Alex031544, 2017-04-02:
Hi,

in our project we found out, that it seems that the signal `SIGINT` isn't been handled and `rclcpp::shutdown()` is not called.

Regards,
Alex

wjwwood, 2017-04-03:
@Alex031544 is your program overriding the signal handler? That, or not ever calling `rclcpp::init()`, is the only reason that would occur.

This line:

https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L64

Gets printed when I send `SIGINT` to any of our demos, like `talker`.

You haven't given enough information for me to help you.

Alex031544, 2017-04-03:
Hello,

here are some simple examples to proof our suspicion:

1. A first example which show an code how we think it shall work:
[image]
   We expect that after `SIGINT` (**Ctrl+C**) `rclcpp::spin(node);` will terminate and two outputs will be visible on the terminal.  
   As you describe the output `signal_handler(2)` from the rclcpp-signal-handler comes up. But the printouts behind spin don't.

2. Now we add an own signal handler and a test class to see how it terminates now:
```
#include <iostream>
#include "rclcpp/rclcpp.hpp"
#include <signal.h>

class SubTest{
public:
	SubTest(){
		std::cout << "SubTest: call of constructor" << std::endl;
	}

	~SubTest(){
		std::cout << "SubTest: call of destructor" << std::endl;
	}
};

class Test : private SubTest{
public:
	Test(){
		std::cout << "Test: call of constructor" << std::endl;
	}

	~Test(){
		std::cout << "Test: call of destructor" << std::endl;
	}
};

void
h_sig_sigint( int signum )
{
	std::cout << "Receive signum: " << signum << std::endl;
	rclcpp::shutdown();
}

int
main (int argc, char * argv[])
{
	std::cout << "test_prog" << std::endl;

	signal(SIGINT, h_sig_sigint);

	// initialize ROS2
	rclcpp::init(argc, argv);
	// initialize a node
	auto node = rclcpp::node::Node::make_shared("test_prog");

	// an test class
	Test test;

	std::cout << "spin" << std::endl;
	rclcpp::spin(node);
	
	printf("start termination");
	
	printf("... done\n");
	return 0;
}
```
and the corresponding output:
[image]
We expect, that our signal handler is overwritten by `rclcpp::init()` but it seems that it don't. For me it is surprising that the output of both callbacks become visible. Further we search for usages of `rclcpp::shutdown()` but we don't find anything in relation to the ROS2 signal handler.

We hope, that the issue now becomes a bit more clear.

Regards,
Alex

wjwwood, 2017-04-03:
@Alex031544 it looks like you're using OpenSplice? Can you try with one of the implementations we currently support like Fast-RTPS or Connext? Which version of ROS 2 are you using? I can't tell you how to change implementations without knowing which version you're using.

I can't think of any reason off hand that our signal handler would not cause `rclcpp::spin()` to exit.

Also, I'd like to easily reproduce your first example, but you put in a picture of text. Please, please, please do not do this. I can't copy it, it cannot be searched, and I cannot quote it.

Alex031544, 2017-04-03:
Yeah sorry for that, but the screenshot was taken yesterday. The sources are gone.

The version of ROS2 on my machine is *release-beta1* and yes we are using opensplice. The reason for that is that we got the best results of stability during our first tests based on *alpha7*.

Here the code:
```
#include <iostream>
#include "rclcpp/rclcpp.hpp"

int
main (int argc, char * argv[])
{

	std::cout << "test_prog" << std::endl;

	// initialize ROS2
	rclcpp::init(argc, argv);
	// initialize a node
	auto node = rclcpp::node::Node::make_shared("test_prog");

	std::cout << "spin" << std::endl;
	rclcpp::spin(node);
	
	printf("start termination");
	
	printf("... done\n");
	return 0;
}
```

Karsten1987, 2017-04-03:
So I just tried this with connext, and it exits just fine on connext and fastrtps:

```
int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);

  auto node = rclcpp::node::Node::make_shared("talker");
  rclcpp::spin(node);
  fprintf(stderr, "exiting..\n");
  return 0;
}
```

Alex031544, 2017-04-03:
hm. But where `rclcpp` is called? And why it doesn't work as expected for opensplice?

wjwwood, 2017-04-03:
> hm. But where rclcpp is called?

I don't know what you're asking here.

> And why it doesn't work as expected for opensplice?

I don't know off-hand. We decided to stop testing and fixing issue with OpenSplice for now:

https://github.com/ros2/ros2/wiki/Alpha8-Overview#changes-to-supported-dds-vendors

However, you can feel free to open issues on the `rmw_opensplice` issue tracker, and if you figure out what the issue with the signal handler is, we'd be happy to help you get a patch merged.

> If it is necessary/advisable to switch to FastRTPS we need good reasons, because this means a lot of tests. In the past FastRTPS we had have some stability problems with the network.

It's not necessary to switch, but we're focusing on maintaining and improving our Fast-RTPS implementation right now but we're not actively working on our OpenSplice implementation right now. So by sticking with OpenSplice you might be taking on some of the extra maintenance burden, but if you find OpenSplice to work better for you, then that might be the right decision even if it means taking on some of the maintenance burden with `rmw_opensplice`. Our intention was to make it so that anyone could maintain or even add implementations.

Alex031544, 2017-04-03:
I don't end my question above:

But where `rclcpp::shutdown` is called? I expected that the signal handler calls them.

wjwwood, 2017-04-03:
`rclcpp::shutdown()` itself is not called from the signal handler, but the same effect is put in motion. For example, this is the code in the signal handler:

https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L87-L98

And this is the code of `rclcpp::shutdown()`:

https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/utilities.cpp#L165-L176

This duplication of code might need to be de-duplicated, or maybe `rclcpp::shutdown()` should be called directly from the signal handler. However, there might have been a reason it is not this way, I don't remember.

Alex031544, 2017-04-03:
I put `shutdown` inside the signal handler but this shows no effect.

The question I ask to myself is why both handlers (ROS2 and our) of the example above are bind to the signal. I thought that only on signal handler can be bound to a signal.

I will do further investigations and leave my results here if I found something new.

wjwwood, 2017-04-03:
> The question I ask to myself is why both handlers (ROS2 and our) of the example above are bind to the signal. I thought that only on signal handler can be bound to a signal.

That is typically the case (always the case in my experience), and that's why the ROS 2 signal handler is careful to store the existing signal handler:

https://github.com/ros2/rclcpp/blob/734ac278dbafe92d13269438cce95635d50e39fe/rclcpp/src/rclcpp/utilities.cpp#L111-L123

and then call it during our own signal handler:

https://github.com/ros2/rclcpp/blob/734ac278dbafe92d13269438cce95635d50e39fe/rclcpp/src/rclcpp/utilities.cpp#L65-L85

allenh1, 2017-06-22:
@wjwwood @dhood is this possibly related to that issue you were experiencing with AMCL last Thursday?

dhood, 2017-06-22:
nah that was a launch issue, resolved by https://github.com/ros2/launch/pull/55 thanks

Karsten1987, 2018-02-26:
@Alex031544 given that we by now call `rclcpp::shutdown` in all of our demos and there is no update on this ticket, I'll take it this issue is resolved.
Thanks for iterating over it and please feel free to re-open it in case you still have trouble with it.

closed 2018-02-26 by Karsten1987

wjwwood, 2018-04-11:
@EwingKang my opinion on this issue is that either:

- you are not returning from a callback in your node
- you are using something that overrides the signal handler for SIGINT but does not call the ROS signal handler afterwards

In either case, I recommend you add console output and/or more of the code that's not working.

EwingKang, 2018-04-11:
Hello, @wjwwood  Thanks for your reply. The following is my OmniIo node. All of the node did return normally as far as I tested (with printf). The destructor is never called like afore mentioned.   
- I'm running this node under `sueo -sE` because it need to access the motherboard GPIO library. 
- Also, all the clock call(clk2, clk3, or even `this-now()`) will fail, they always stuck at the same number, e.g. "140735253311824" where the system chrono gives the correct, advancing clock like "1523409062194556622".   
Not sure if the information is related or not Hope this helps.

```
#include <stdio.h>
#include "OmniIoNode.hpp"

using namespace std::chrono_literals;

/* * * * * * * * * * 
 * Private Methods *
 * * * * * * * * * */
void OmniIoNode::topic_callback(const std_msgs::msg::String::SharedPtr msg)
{
    // Print the received message
    printf("------------------------------------------------------------------\n");
    printf("=>>> receive from -- Topic <\"%s\">: \"%s\".\n", TOPIC_CMD, msg->data.c_str());
    printf("\n");
    
    // do some hardware related stuff
    return;
}

void OmniIoNode::timer_callback()
{
    // read contact switch
    uint32_t contact_sw_level, onoff_sw_level;
    gpio_sw_contact_->ReadLevel(contact_sw_level);
    gpio_sw_onoff_->ReadLevel(onoff_sw_level);
    
    /*rclcpp::TimeSource ts(shared_from_this());
    rclcpp::Clock::SharedPtr clk2 = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);
    ts.attachClock(clk2);
    rclcpp::Clock clk3;*/
    
    uint32_t level[4] = {0};
    if( ***detecting_rising_edge*** )
    {
        //RCLCPP_INFO(this->get_logger(), "CONTACT!!! time: %ld", clk2->now());
        RCLCPP_INFO(this->get_logger(), "CONTACT!!! time: %ld", this->now());
        RCLCPP_INFO(this->get_logger(), "CONTACT!!! time: %ld", std::chrono::system_clock::now());       
        // set LED output
    }
    return;
}

void OmniIoNode::set_led(const uint32_t (&state)[4])
{
    //set gpio output
    return;
}

/* * * * * * * * * * 
 * Public Methods *
 * * * * * * * * * */
 //====== Constructor ======//
OmniIoNode::OmniIoNode() : Node("omni_gpio")
{
    subscription_ = this->create_subscription<std_msgs::msg::String>(
            TOPIC_CMD, std::bind(&OmniIoNode::topic_callback, this, _1),
            rmw_qos_profile_sensor_data);
    timer_ = this->create_wall_timer(10ms, std::bind(&OmniIoNode::timer_callback, this));
    clock_ = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);

    //hardware gpio library initial set-up
  }

//====== Destructor ======//
OmniIoNode::~OmniIoNode()
{
    uint32_t level[4] = {0};
    set_led(level);
    //gpio ending setup
    printf("Node shutting down, reset all GPIOs");
    //destruct GPIO library
}

EwingKang, 2018-04-11:
I've found out how to trigger my issue. It turns out if I link a perticular library within the `add_executable()` in CMakeLists.txt, the signal_handler() sould stop working, i.e. showing only `signal_handler(2)` but no return of spin(node)   

I guess it kinda make sense since the library is about motherboard hardware(GPIO, I2C, sensor etc) stuff, so it probably mess with linux kernal. What bothers me is that it doesn't matter if I've include/use the library in my node source or not, it'll affect all the same if it's add in the CMakeLists. Sadly a quick text search of `SIGINT` or `signal` returns nothing so I may nevery know what's wrong inside that library.

Leo-Nogismus, 2019-10-31:
Sorry to bring this up again after such a long time, I have the same issue using ros dashing with Connext as far as I know. Should I open a new issue or just keep posting here?