ros2/rclcpp #318: rclcpp lifecycle: what is the design motivation for separate lifecycle_node and node classes?

satyan, 2017-04-06:
I was trying to use lifecycle_node for making managed nodes and discovered that they don't extend Node class. It seems like composition is preferred, which is fine. But, wanted to learn what's the design motivation to create a parallel structure for the lifecycle strategy.

Appreciate the response.

wjwwood, 2017-04-07:
> But, wanted to learn what's the design motivation to create a parallel structure for the lifecycle strategy.

We have a node class which has lifecycle and one that does not because we thought it made sense to have lifecycle be an addition on top of the "core" node functionality. It's a parallel structure because we wanted to avoid inheritance in favor of composition (as you noted already).

In the future it's possible that the plain `rclcpp::Node` class will be one which has lifecycle features within it, and in that case we'd probably have a `rclcpp::nodes::BasicNode`, or something like that, which does not.

@Karsten1987 might have some input too.

I'm going to close this, since otherwise these types of question tickets tend to stay open forever. But, you can feel free to continue the conversation after it is closed. I can reopen it if necessary.

closed 2017-04-07 by wjwwood

Karsten1987, 2017-04-07:
Yeah, @wjwwood basically answered the question already. We decided to favor composition mainly of being able to test/mock the individual components of the node class. 

As you might have noticed, we tried to keep the basic interface / public member function of the lifecycle node as similar as the original node class. This is a) for not breaking the API too much and b) for being able to migrate the lifecycle components into the node class.