ros2/rclcpp #319: 100% processor utilisation if timer is canceled

ArkturusMengsk, 2017-04-07:
As soon as a wall timer is canceled, the processor load will grow immediately to 100%.

`myTimer = this->create_wall_timer(1s, std::bind(&TestNode::timeout, this));`
`myTimer->cancel();`

dirk-thomas, 2017-04-07:
Please provide a complete example (see http://sscce.org).

ArkturusMengsk, 2017-04-07:
That is nearly all, that is needed to reproduce the described behavior.

main.cpp
```
#include <rclcpp/rclcpp.hpp>
#include "testnode.h"
int main( int argc, char *argv[] ) {
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<TestNode>());
  return 0;
}
```

testnode.h
```
#ifndef TESTNODE_H
#define TESTNODE_H
#include "rclcpp/rclcpp.hpp"
class TestNode : public rclcpp::Node {
  public:
    TestNode();
  private:
    void timeout();
    rclcpp::TimerBase::SharedPtr  myTimer;
};
#endif
```

testnode.cpp
```
#include <chrono>
#include "testnode.h"
using namespace std::chrono_literals;
TestNode::TestNode() : Node("TestNode") {
  myTimer = this->create_wall_timer(1s, std::bind(&TestNode::timeout, this));
  myTimer->cancel();
}
void TestNode::timeout() {  printf("timeout\n"); }
```

mikaelarguedas, 2017-04-26:
@ros2/team please try to reproduce and comment here

mikaelarguedas, 2017-05-10:
@wjwwood to see if recent changes fixed this

wjwwood, 2017-09-01:
@ArkturusMengsk I was able to reproduce this (thanks for the minimal example!) and we came up with a fix in https://github.com/ros2/rcl/pull/163.

Thanks again for helping us narrow it down, and sorry it took us so long to get to it.

mikaelarguedas, 2017-09-08:
I'm going to close thins now that ros2/rcl#163 has been merged. Thanks again @ArkturusMengsk for the original report and the reproducible example!

closed 2017-09-08 by mikaelarguedas