ros2/rclcpp #324: TestStateMachineInfo.available_transitions test flaky

dhood, 2017-04-26:
This test has failed on 93 out of the past 100 of the linux "repeated" nightly jobs.

e.g. http://ci.ros2.org/job/nightly_linux_repeated/624/testReport/junit/(root)/TestStateMachineInfo/available_transitions/

It has also failed twice on OSX (jobs 603 and 622, both mini2). Is it possible there's a race condition?

Karsten1987, 2017-04-27:
The lifecycle implementation needs a general overlook. There are no rmw_allocators used and some functions definitely deserve a "__attribute__((warn_unused_result))" .

Do you have an idea when this flakiness started?

dhood, 2017-04-27:
since Dec 15 (http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/499/) which I think is when it was initially added

Karsten1987, 2017-06-04:
I believe that can be closed, right?

dhood, 2017-06-05:
Could you reference the changes that would have fixed it? It did still fail [11 days ago](http://ci.ros2.org/view/nightly/job/nightly_linux_repeated/663/), but maybe the changes are more recent

Karsten1987, 2017-06-05:
I believe this fixed it:
https://github.com/ros2/rclcpp/pull/327

It was committed 11days ago :)

dhood, 2017-06-05:
looks like it worked then!

closed 2017-06-05 by dhood