ros2/rclcpp #329: Unexpected behavior of rclcpp_lifecycle::Transition.label() and rclcpp_lifecycle::State.label()

breh, 2017-05-26:
I'm playing with state transition and trying to use Transition and State objects. Unfortunately I'm seeing weird behaviors when it comes to dealing with labels in those objects. 

Looks like the way how label strings are handled is not correct - the constructor takes a string reference and then it passes the pointer to the underlying string using c_str() to the actual handle holding the label. Unfortunately since the constructor parameter is a reference and the constructor only assigns a pointer to the underlying char* buffer, it might cause issues when the original string gets deleted / destroyed and thus the label being corrupted in the underlying structure. 

I believe the proper solution should be to copy the actual characters instead of just assigning the pointer.

Karsten1987, 2017-05-26:
Thanks for reporting it, I ran into it earlier this week and hopefully patched it in https://github.com/ros2/rclcpp/pull/327

This is going to be merged hopefully soon.

breh, 2017-05-26:
Great, thanks. Will try to verify tomorrow or next week :).

Karsten1987, 2017-06-04:
@breh Did that work for you?

Karsten1987, 2017-06-07:
I am closing this, assuming it worked out for you as well.
Please feel free to re-open or post comment on this ticket.

closed 2017-06-07 by Karsten1987