ros2/rclcpp #335: support for AsyncSpinner

Karsten1987, 2017-06-05:
As we currently don't have the ROS2 equivalent for the AsyncSpinner, I helped myself out by excluding the `executor::spin()` function and running this in a separate thread.
Is that the recommended way and how such an asyncspinner could look like in the ROS2 world?

```
void
spin(rclcpp::executors::multi_threaded_executor::MultiThreadedExecutor * exe)
{
    exe->spin();
}

// main

rclcpp::executors::multi_threaded_executor::MultiThreadedExecutor executor;
auto future_handle = std::async(std::launch::async, spin, &executor);
```

wjwwood, 2017-06-05:
This was already asked in the forum (at least I think it is a duplicate):

https://discourse.ros.org/t/async-executor-in-ros2/1575

I'd recommend what I put in the discourse thread or this:

```cpp
using rclcpp::executors::multi_threaded_executor::MultiThreadedExecutor;
MultiThreadedExecutor executor;
// executor.add_node(...);
std::thread executor_thread(std::bind(&MultiThreadedExecutor::spin, &executor));
```

Karsten1987, 2017-06-05:
great. thanks for the link. I didn't see that.

closed 2017-06-05 by Karsten1987

itfanr, 2019-06-26:
For ros2 dashing:
```
     using rclcpp::executors::MultiThreadedExecutor;
    MultiThreadedExecutor executor;
    executor.add_node(xxx_node);
```

pankhurivanjani, 2019-07-21:
Hello @itfanr !
I am having issues in implementing it in ROS dashing.

With ROS1 I was using start and stop. What to do in this case?

Also if I have:

```
lc2->start();
std::cout << "hi";
 std::cout << "hey"; ```

where start:
```
void start(){
 RCLCPP_INFO(this->get_logger(), "starting");
this->executor->spin();
}```

My code is not printing "hi" and "hey". It is not moving to further lines wheras with ROS since I used asyncspinner I had no problem.