ros2/rclcpp #343: function_traits - decltype cannot resolve address of overloaded function with gcc 7

jpgr87, 2017-07-19:
I'm building ros2 beta2, and I've run into some issues with some code in function_traits and its tests.

* fedora 26 x86_64
* gcc-7.1.1

```
In file included from /home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:20:0:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp: In instantiation of ‘struct rclcpp::function_traits::function_traits<std::_Bind<int (ObjectMember::*(ObjectMember*, std::_Placeholder<1>))(bool)> >’:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:394:72:   required from here
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:52:85: error: decltype cannot resolve address of overloaded function
       typename function_traits<decltype( & FunctionT::operator())>::arguments>::type;
                                                                                     ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:57:72: error: decltype cannot resolve address of overloaded function
   using argument_type = typename std::tuple_element<N, arguments>::type;
                                                                        ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:59:96: error: decltype cannot resolve address of overloaded function
   using return_type = typename function_traits<decltype( & FunctionT::operator())>::return_type;
                                                                                                ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp: In member function ‘virtual void TestFunctionTraits_argument_types_Test::TestBody()’:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:395:5: error: template argument 2 is invalid
     >::value, "Functor accepts a bool as first argument");
     ^
In file included from /home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:20:0:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp: In instantiation of ‘struct rclcpp::function_traits::function_traits<std::_Bind<int (ObjectMember::*(ObjectMember*, std::_Placeholder<1>, std::_Placeholder<2>))(bool, bool)> >’:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:404:73:   required from here
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:52:85: error: decltype cannot resolve address of overloaded function
       typename function_traits<decltype( & FunctionT::operator())>::arguments>::type;
                                                                                     ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:57:72: error: decltype cannot resolve address of overloaded function
   using argument_type = typename std::tuple_element<N, arguments>::type;
                                                                        ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:59:96: error: decltype cannot resolve address of overloaded function
   using return_type = typename function_traits<decltype( & FunctionT::operator())>::return_type;
                                                                                                ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:405:5: error: template argument 2 is invalid
     >::value, "Functor accepts a bool as first argument");
     ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:411:5: error: template argument 2 is invalid
     >::value, "Functor accepts a bool as second argument");
     ^
In file included from /home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:20:0:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp: In instantiation of ‘struct rclcpp::function_traits::function_traits<std::_Bind<int (ObjectMember::*(ObjectMember*, std::_Placeholder<1>, std::_Placeholder<2>))(bool, float)> >’:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:422:8:   required from here
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:52:85: error: decltype cannot resolve address of overloaded function
       typename function_traits<decltype( & FunctionT::operator())>::arguments>::type;
                                                                                     ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:57:72: error: decltype cannot resolve address of overloaded function
   using argument_type = typename std::tuple_element<N, arguments>::type;
                                                                        ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/include/rclcpp/function_traits.hpp:59:96: error: decltype cannot resolve address of overloaded function
   using return_type = typename function_traits<decltype( & FunctionT::operator())>::return_type;
                                                                                                ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:423:5: error: template argument 2 is invalid
     >::value, "Functor accepts a bool as first argument");
     ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:431:5: error: template argument 2 is invalid
     >::value, "Functor accepts a float as second argument");
     ^
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp: In member function ‘virtual void TestFunctionTraits_check_arguments_Test::TestBody()’:
/home/rich/code/ros2-beta2/src/ros2/rclcpp/rclcpp/test/test_function_traits.cpp:562:78: error: ‘value’ is not a member of ‘rclcpp::function_traits::check_arguments<std::_Bind<int (ObjectMember::*(ObjectMember*, std::_Placeholder<1>))(bool)>, bool>’
     rclcpp::function_traits::check_arguments<decltype(bind_one_bool), bool>::value,
                                                                              ^~~~~
make[2]: *** [CMakeFiles/test_function_traits.dir/build.make:63: CMakeFiles/test_function_traits.dir/test/test_function_traits.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1436: CMakeFiles/test_function_traits.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

allenh1, 2017-07-20:
Let me look into this -- I have GCC 8 on my laptop. let me see if I can reproduce this, or if I need to talk to the GCC people again. ;)

allenh1, 2017-07-20:
Full compiler error:

```
Process package 'composition' with context:
--------------------------------------------------------------------------------
 source_space => /home/allenh1/ros2_ws/src/ros2/demos/composition
  build_space => /home/allenh1/ros2_ws/build/composition
install_space => /home/allenh1/ros2_ws/install
   make_flags => -j8, -l8
  build_tests => False
--------------------------------------------------------------------------------
+++ Building 'composition'
Running cmake because arguments have changed.
==> '. /home/allenh1/ros2_ws/build/composition/cmake__build.sh && /usr/bin/cmake /home/allenh1/ros2_ws/src/ros2/demos/composition -DBUILD_TESTING=0 -DAMENT_CMAKE_SYMLINK_INSTALL=1 -DPYTHON_INSTALL_DIR=install/lib/python3.5/site-packages -DCMAKE_CXX_COMPILER=g++ -DCMAKE_C_COMPILER=gcc -DCMAKE_CXX_STANDARD=14 -DCMAKE_INSTALL_PREFIX=/home/allenh1/ros2_ws/install' in '/home/allenh1/ros2_ws/build/composition'
-- The C compiler identification is GNU 8.0.0
-- The CXX compiler identification is GNU 8.0.0
-- Check for working C compiler: /home/allenh1/bin/gcc
-- Check for working C compiler: /home/allenh1/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/allenh1/bin/g++
-- Check for working CXX compiler: /home/allenh1/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found ament_cmake: 0.0.2 (/home/allenh1/ros2_ws/install/share/ament_cmake/cmake)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found ament_index_cpp: 0.0.2 (/home/allenh1/ros2_ws/install/share/ament_index_cpp/cmake)
-- Found class_loader: 1.0.0 (/home/allenh1/ros2_ws/install/share/class_loader/cmake)
-- Found example_interfaces: 0.0.2 (/home/allenh1/ros2_ws/install/share/example_interfaces/cmake)
-- Found rclcpp: 0.0.2 (/home/allenh1/ros2_ws/install/share/rclcpp/cmake)
-- Found rmw_implementation_cmake: 0.0.2 (/home/allenh1/ros2_ws/install/share/rmw_implementation_cmake/cmake)
-- Using RMW implementation 'rmw_fastrtps_cpp' as default
-- Found rmw_fastrtps_cpp: 0.0.2 (/home/allenh1/ros2_ws/install/share/rmw_fastrtps_cpp/cmake)
-- Found fastrtps_cmake_module: 0.0.2 (/home/allenh1/ros2_ws/install/share/fastrtps_cmake_module/cmake)
-- Found FastRTPS: /home/allenh1/ros2_ws/install/include  
-- Found rosidl_default_generators: 0.0.2 (/home/allenh1/ros2_ws/install/share/rosidl_default_generators/cmake)
-- Found std_msgs: 0.0.2 (/home/allenh1/ros2_ws/install/share/std_msgs/cmake)
-- Using single matching rosidl_typesupport_c: rosidl_typesupport_introspection_c
-- Using single matching rosidl_typesupport_cpp: rosidl_typesupport_introspection_cpp
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.3", minimum required is "3.5") 
-- Found python_cmake_module: 0.0.2 (/home/allenh1/ros2_ws/install/share/python_cmake_module/cmake)
-- Found PythonLibs: /usr/lib64/libpython3.5m.so (found suitable version "3.5.3", minimum required is "3.5") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.5m
-- Using PYTHON_LIBRARIES: /usr/lib64/libpython3.5m.so
-- Found PythonExtra: .so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/allenh1/ros2_ws/build/composition
==> '. /home/allenh1/ros2_ws/build/composition/cmake__build.sh && /usr/bin/make -j8 -l8' in '/home/allenh1/ros2_ws/build/composition'
Scanning dependencies of target composition__cpp
Scanning dependencies of target dlopen_composition
[  1%] Generating C code for ROS interfaces
Scanning dependencies of target listener_component
Scanning dependencies of target composition__py
Scanning dependencies of target server_component
Scanning dependencies of target client_component
Scanning dependencies of target talker_component
[  3%] Generating C++ code for ROS interfaces
[  5%] Generating Python code for ROS interfaces
[  7%] Building CXX object CMakeFiles/server_component.dir/src/server_component.cpp.o
[  9%] Building CXX object CMakeFiles/listener_component.dir/src/listener_component.cpp.o
[ 11%] Building CXX object CMakeFiles/dlopen_composition.dir/src/dlopen_composition.cpp.o
[ 13%] Building CXX object CMakeFiles/client_component.dir/src/client_component.cpp.o
[... 83 lines cut ...]
make[1]: *** [CMakeFiles/Makefile2:640: CMakeFiles/client_component.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 30%] Building CXX object CMakeFiles/composition__rosidl_typesupport_introspection_cpp.dir/rosidl_typesupport_introspection_cpp/composition/srv/load_node__response__type_support.cpp.o
make[2]: *** [CMakeFiles/talker_component.dir/build.make:63: CMakeFiles/talker_component.dir/src/talker_component.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:603: CMakeFiles/talker_component.dir/all] Error 2
Scanning dependencies of target composition__rosidl_typesupport_introspection_c
[ 32%] Building C object CMakeFiles/composition__rosidl_typesupport_introspection_c.dir/rosidl_typesupport_introspection_c/composition/srv/load_node__type_support.c.o
[... cut ...]
[ 46%] Linking CXX shared library liblistener_component.so
[ 46%] Built target listener_component
[ 46%] Built target composition__py
make: *** [Makefile:128: all] Error 2

<== Command '. /home/allenh1/ros2_ws/build/composition/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/allenh1/ros2_ws/build/composition' with exit code '2'
<== Command '. /home/allenh1/ros2_ws/build/composition/cmake__build.sh && /usr/bin/make -j8 -l8' failed in '/home/allenh1/ros2_ws/build/composition' with exit code '2'
```

jpgr87, 2017-07-20:
Yes, I was building beta2 following the instructions [here](https://github.com/ros2/ros2/wiki/Linux-Development-Setup)

The command I used was:
```
src/ament/ament_tools/scripts/ament.py build --build-tests --symlink-install
```

mikaelarguedas, 2017-07-26:
@allenh1 to provide feedback on this and move in the appropriate column

clalancette, 2017-07-28:
I think this is originally @esteve 's code. @esteve , any thoughts on what is going on here and how to fix it?

allenh1, 2017-07-28:
I'm currently talking to the GCC people to find out if this is a bug. I'll post back when I know more.

esteve, 2017-07-29:
@clalancette thanks for reaching out. I've submitted https://github.com/ros2/rclcpp/pull/346 which should address this.

@allenh1 it's actually an issue in `function_traits`. `function_traits` is rather fragile when dealing with `std::bind` because there's no requirement in the C++ standard on how to implement it, so `function_traits` depends on the internal implementation to match the return type, arguments, etc. of a functor. There's nothing much that can be done, except updating `function_traits` whenever `std::bind` changes, which is not very often.

closed 2017-08-01 by allenh1

mikaelarguedas, 2017-08-01:
Closed via #346 thanks @jpgr87 for the initial report