ros2/rclcpp #349: resetting subscriber leads to crash

Karsten1987, 2017-08-04:
given the following code: https://gist.github.com/Karsten1987/82837b87fd11ef23353754b9fa31cf7a

the situation:

1.) node spawns a subscriber and a publisher
2.1) main thread spins the executor, having sub and pub talking
2.2) second thread randomly stops the subscription (resetting the shared_ptr of it)
3.) program crashes with the backtrace at the bottom of the gist.

quickly discussed the problem with @wjwwood, when resetting the subscription, the executor still has a handle on it which leads to race conditions. The problem can be addressed by turning the handle into a shared pointer.

dhood, 2017-08-23:
this is the permalink to the function @Karsten1987 is referencing in his post (in case those lines move around): https://github.com/ros2/rclcpp/blob/b90676871d397cbfdcfecdf265869efb0bd97d9f/rclcpp/include/rclcpp/subscription.hpp#L81-L83

deng02, 2018-01-16:
We have been hitting this issue when using the ROS2-to-ROS1 ros1_bridge with bridge_all_2to1_topics set to false.  When the ROS1 subscriber for a topic disappers the ROS2 subscriber is destroyed and sometimes this happens while the wait set is still holding a raw pointer to a member of that object.  The result is various segfault errors within rmw_wait.

wjwwood, 2018-03-12:
I believe https://github.com/ros2/rclcpp/pull/431 closes this, please comment here if this is still an issue.

closed 2018-03-12 by wjwwood