ros2/rclcpp #360: Expose timeout option for parameter service calls by SyncParametersClient

dhood, 2017-08-23:
As an example, [SyncParametersClient::set_parameters](https://github.com/ros2/rclcpp/blob/40b09b5b14c6df9fe9eaf6a6a907c65c5b70747e/rclcpp/src/rclcpp/parameter_client.cpp#L324..L337) doesn't take a timeout option. So, if anything goes wrong in the service call (e.g. the server goes down), we will get stuck waiting indefinitely.

Exposing a timout that is set to -1 by default (as is done for [SyncParametersClient::wait_for_service](https://github.com/ros2/rclcpp/blob/124500511bb086b869876108d5d0b61511b3227c/rclcpp/include/rclcpp/parameter_client.hpp#L231)) would be appropriate.

Parameter tests in `test_rclcpp` (e.g. [this one](https://github.com/ros2/system_tests/blob/67efe8e319b997c389c266d9ac237b850807ba4e/test_rclcpp/test/test_local_parameters.cpp#L100)) should be updated to specify a timeout, so that the other tests in the file can be run even if one of them hangs.

wjwwood, 2020-03-19:
Updated link:

https://github.com/ros2/rclcpp/blob/96ebf59a6045a535730d98fff25e522807c7aa75/rclcpp/src/rclcpp/parameter_client.cpp#L453-L468

fujitatomoya, 2020-12-22:
@wjwwood 

thanks for the review, i think we can close this.

closed 2020-12-23 by wjwwood