ros2/rclcpp #370: gcov Integration for gtest Coverage

andrew-konecki-apexai, 2017-09-05:
I was wondering if there were any metrics with the use of gtest within the rclcpp package?  I am specifically trying to determine code coverage that gtest is providing with respect source of the package with say gcov.  

At the moment I am attempting to figure out where I would need to include the compilation options to support this.  Is there a place within the Ament system that would provide this with respect to a specific package?

Thanks,

Andrew

dirk-thomas, 2017-09-05:
`ament_tools` doesn't offer a way to specify package specific options. A [new build tool](http://design.ros2.org/articles/build_tool.html) should definitely offer that capability.

Currently we don't have a way to compute such metrics. While it would certainly be nice to have this hasn't made it up on the list of priorities in the past.

Karsten1987, 2018-02-26:
@andrew-konecki-apexai does this answer your question? If so, feel free to close this issue or re-iterate over it in case there are some more open questions.

dirk-thomas, 2018-02-26:
Based on the time since this was filled and that there was no response since then I will close this for now. Please feel free to comment and it can be considered to be reopened.

closed 2018-02-26 by dirk-thomas