ros2/rclcpp #377: Support for array parameters

ayrton04, 2017-09-23:
I recently posted a [question](https://discourse.ros.org/t/array-parameters/2607/2) on the Discourse site about support for array parameters for primitive data types. Unless there are any specific objections to me doing so, I'd like to submit a PR to enable this functionality.

wjwwood, 2017-09-25:
@ayrton04 it would be nice to first have a pull request that modifies the design doc for parameters which addresses issues like what to call the types, how they will behave, how it might impact the API, or other considerations. Also it should then remove this section 😄:

http://design.ros2.org/articles/ros_parameters.html#support-for-arrays-of-primitives

After that we'll have an agreed on approach, then the implementation becomes easier to do and review and get merged.

----

For the name, I think your suggested `PARAMETER_*_ARRAY` sounds fine to me.

----

Also, just so that you are aware, we have been planning to restructure the parameter code so that more is shared in between C++ and Python by moving some of the code that currently exists in our C++ API (`rclcpp`) to our common C API (`rcl`).

I don't think this should prevent you from working on this, but it might be something we have to merge and/or resolve conflicts with.

ayrton04, 2017-09-26:
@wjwwood Yes, good call. I'll start with PRs to update the design documentation.

Pushing shared functionality into `rcl` is a great idea. I'll admit to having already started on the new array data type parameters, and it's all been pretty straightforward so far (especially since `BYTES` was already implemented), so hopefully resolving conflicts when the core functionality moves to `rcl` won't be too onerous either.

mikaelarguedas, 2018-06-06:
Closing this as https://github.com/ros2/design/pull/152 has been merged. Thanks again @ayrton04 for the contribution!

closed 2018-06-06 by mikaelarguedas