ros2/rclcpp #385: 'change_state' svc cb in LifecycleNode::init(..) not set(..) with std::move(..)?

gavanderhoorn, 2017-10-13:
This is more a question than an issue: all callbacks for the lifecycle management services are `std::move(..)`ed in the `AnyServiceCallback<..>::set(..)` calls in `LifecycleNodeInterfaceImpl::init(..)`, except the one for `change_state` ([here](https://github.com/ros2/rclcpp/blob/2c6d95946eb852fb1fa6da7e81d4b30a9e7a7ad3/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L100)).

It's probably my understanding of the code (or lack thereof at the moment), but I can't really figure out why that is.

Is this an oversight / copy-pasta?

Karsten1987, 2017-10-14:
thanks @gavanderhoorn for catching that. Related PR can be found here: https://github.com/ros2/rclcpp/pull/387

closed 2017-10-16 by Karsten1987

gavanderhoorn, 2017-10-16:
Good to know I'm not 'crazy' :)

👍 @Karsten1987.