ros2/rclcpp #392: Single-threaded executor can be starved by timers

dhood, 2017-11-08:
**Context** (comes from https://github.com/ros2/demos/pull/187): A single-threaded executor has a timer scheduled for every N seconds, and also a service server.

**Bug**: If the timer callback takes >=N seconds to complete, the executor will never process service requests once the timer is triggered for the first time.

This is because of the combination of the following:

1. When the spinning executor calls to `get_next_executable`, [it does _not_ call `rcl_wait` since the timer's ready](https://github.com/ros2/rclcpp/blob/070b3125c1a7f76fbde3e31973d89a052a2ea652/rclcpp/src/rclcpp/executor.cpp#L546..L550), so the service is not marked as ready.
1. Even if you force `get_next_executable` to always call `wait_for_work` so the server can get receive its request, the server will still _not_ get processed by the executor because [the timer will always get chosen first as the `next_ready_executable`](https://github.com/ros2/rclcpp/blob/070b3125c1a7f76fbde3e31973d89a052a2ea652/rclcpp/src/rclcpp/executor.cpp#L517..L521).

Forcing `get_next_executable` to always call `wait_for_work` _and_ giving timers lower priority in `get_next_ready_executable` will fix this situation, but it is inefficient to wait when it's not necessary, and I'm not sure checking timers last is a fix-all (could there be a parallel situation when the server needs to be the lowest priority?).

@dirk-thomas mentioned that a queue of some sort is probably more appropriate, so that `get_next_executable` processes events in the order that they were received.

For now, we have to recommend that users not permit timer callbacks to block for longer than the duration at which they're scheduled (this might be in our documentation somewhere already?).

mikaelarguedas, 2017-11-09:
related to executor starvation reported https://github.com/ros2/rclcpp/issues/280
Right now we will match the behavior of rclpy and provide documentation as of how to make custom executors.

ubald, 2019-03-12:
:wave: I ended up here trying to find our why when I had multiple timers with short intervals (and longer callbacks) only the first one worked, the others are never executed. 

Is it possible that both issues are related?

wjwwood, 2019-03-12:
@ubald I do not believe what you've described is the issue here. This issue is essential that we do not have fair scheduling and so as long as timers are always ready then other things like Services will never get handled. However, more than "the first one" work in this case.

ubald, 2019-03-12:
Thanks for the quick reply!

Please bear with me here, I just started with ROS2 yesterday, I don't have a full grasp of all the pieces involved yet, but when looking at the code it looks to me like it's related:

When calling `get_next_timer`: 
https://github.com/ros2/rclcpp/blob/070b3125c1a7f76fbde3e31973d89a052a2ea652/rclcpp/src/rclcpp/executor.cpp#L518
It iterates over all timers in the same order each time, so it will always start by checking the first one created https://github.com/ros2/rclcpp/blob/070b3125c1a7f76fbde3e31973d89a052a2ea652/rclcpp/src/rclcpp/executor.cpp#L500-L502
Then if it has been running for longer than it's interval will always be ready [here in rcl](https://github.com/ros2/rcl/blob/1a34301995a29cd3ce3be182613bd6d65176b9cf/rcl/src/rcl/timer.c#L297)

I'll see if I can switch from a debian install to a source build, this way I might try to fix it and make a PR.